BeamNG/BeamNGpy #2: Change camera settings

alessiogambi, 2018-06-30:
A programmatic way to configure how to grab the frames from BeamNG (camera position, FOV, etc) would be ideal.

tdev, 2018-06-30:
z.B. im Lua code:

`
      core_camera.setByName(0, "relative")
      core_camera.proxy_Player('setupRelative',
        vec3(0, 0, 1), -- pos
        vec3(0, 0, 0), -- euler rotation
        90 -- FOV
      )
`

alessiogambi, 2018-07-01:
I updated the Lua file inside the scenario as follows, and got what I needed:
```
     core_camera.setByName(0, "relative")
     core_camera.proxy_Player('setupRelative',
        vec3(-0.5, 2, 0.5), -- pos: more or less in front of the car, on the driver side, slightly above the ground
        vec3(180, 0, 180), -- euler rotation: somehow the default camera is facing backward and flipped upside down
        90 -- FOV
      )
```

Signaltonsalat, 2018-07-02:
Added a [relative_camera](https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.beamng.BeamNGPy.relative_camera) method to set up a camera realtive to the vehicle.

closed 2018-07-02 by Signaltonsalat