BeamNG/BeamNGpy #20: Camera sensor should not show red light

alessiogambi, 2019-01-19:
Images collected using the camera sensor while the car is braking shows a red shadow, despite the camera is positioned in front on the car facing the road.

alessiogambi, 2019-06-12:
Any news on this? I noticed that also the rear-light (white) as visible from the cameras

alessiogambi, 2019-06-12:
Also if a LiDaR sensor is attached you can see the green stripes the sensor on each colored image collected with the camera sensor

Signaltonsalat, 2019-06-19:
The residual flare from the head-/taillights is fixed in the most recent .research release. More work needs to be done to address the issue of the Lidar points, but for now I've added a workaround to disable the Lidar visualization, as it is mostly cosmetic anyway. You can do so by setting `visualized=False` in the constructor.

https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.sensors.Lidar

closed 2019-06-19 by Signaltonsalat