BeamNG/BeamNGpy #21: BeamNGpy should provide a DistanceToGoal sensor

alessiogambi, 2019-02-06:
For checking that the ego car gets to destination having a sensor which retrieve the relative position of the car w.r.t. a given way point might be useful to have.

Signaltonsalat, 2019-02-19:
I would prefer adding functions to query the position/rotation/etc. of objects in a scenario which would in turn allow you to calculate that distance easily by comparing vehicle and goal positions -- that's more powerful than a simple sensor that's tied to a waypoint. Would that work?

alessiogambi, 2019-02-20:
Sure. Probably a specific API to query what elements are present in the scenario works as well.

Signaltonsalat, 2019-06-19:
I've added methods to query a scenario for objects of certain types in the world which return the position, rotation, etc. You can use these and compare them to the vehicle's position.

Here's one of the functions you can use to find waypoints, for example: https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.scenario.Scenario.find_waypoints

And an example of measuring distance between the vehicle and a static object in the scene can be found here: https://github.com/BeamNG/BeamNGpy/blob/master/examples/object_placement.ipynb

closed 2019-06-19 by Signaltonsalat