BeamNG/BeamNGpy #25: The set of triples in the list does not work for poly-line.

tariqsameed, 2019-06-24:
The BeamNG throws an error when I provide the set of triples in the list for the vehicle. 

`try:

        bng.load_scenario(scenario)
        bng.start_scenario()
        line = [(0,0,0.163609),(25,300, 0.163609)]
        vehicle.ai_set_line(line=line)
        vehicle.ai_set_speed(6.944, mode='set')
        vehicle.ai_drive_in_lane(True)
        input('Press enter when done...')
    finally:
        bng.close()`

The sample output in the console is attached. 

[image]

It will be great if you can correct me if I am doing something wrong.

Signaltonsalat, 2019-06-25:
Seems like I forgot to update the docstring for that function, my bad. However, I recommend using the more powerful `ai_set_script` function which uses [BeamNG.drive's ScriptAI](https://www.youtube.com/watch?t=27s&v=11c2lUiPKqw) for driving around. The main difference to `_line` is that, instead of position and *speed*, you define a line by a position and a target *time*, which also allows you to specify lines where the vehicle stops at a certain point — something not really possible with `ai_set_line`. There's an example of how to use `ai_set_script` here: https://github.com/BeamNG/BeamNGpy/blob/master/examples/ai_line.py

tariqsameed, 2019-06-25:
Actually, I want to drive the vehicle between the polyline at a constant speed.  It seems to me that `ai_set_line` will be the most appropriate solution in my case.  I don't want to define the time to reach the destination point. I believe that it should be automatically handled by BeamNG itself.  if possible, please update the documentation of `ai_set_line` so I can run both examples of `ai_set_script.` and `ai_set_line`  What are the changes you have made it `ai_set_line`.

Signaltonsalat, 2019-06-25:
The documentation is already updtaed: https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.vehicle.Vehicle.ai_set_line
There is also the old version of the example above: https://github.com/BeamNG/BeamNGpy/blob/c6215295a942f00388062d8525f20e7fc59159d1/examples/ai_line.py

To clear your vehicle inputs, you can simply send 0 for the respective values: `vehicle.control(throttle=0, steering=0, brake=0, parkingbrake=0)

tariqsameed, 2019-06-25:
The `ai_set_script.` does not give you control over speed. Let suppose I want to drive the car from node A to node B at a constant speed of 35 Kph. I have no way to set the speed because of the time is being set in the ai_script. Secondly, the initial speed of the car at starting waypoint starts from 0. The BeamNG should let us start the car at the speed of 10 kph

closed 2019-06-25 by tariqsameed

lsm2842035890, 2024-03-25:
[image]
[image]
 hello friends~I loaded the gridsmall map and paved a simple road. I want the vehicle to run according to my trajectory. Why does the vehicle not move?