BeamNG/BeamNGpy #28: BeamNG.research Connection problem

autonomobil, 2019-08-17:
I just downloaded (via SVN) the [BeamNG.research repository](https://projects.beamng.com/projects/research/repository). 
[https://projects.beamng.com/projects/research/files](https://projects.beamng.com/projects/research/files) is empty. In a conda enviroment I installed BeamNGpy via `pip install beamngpy`.

When I try the example code with west_coast_usa, BeamNG starts but after a short time it threws the following error:

`055.319|D|GELua.scenario_scenarios.scenarios|should run: false; state: running                                          055.322|E|GELua.core_levels.levels|info.json missing: /levels/garage/info.json                                          055.325|E|GELua.core_levels.levels|No entry point for level found: /levels/garage. Ignoring level.                      055.442|E|GELua.ResearchCom|Error reading from socket: closed                                                           055.472|E|GELua.core_levels.levels|info.json missing: /levels/garage/info.json                                          055.473|E|GELua.core_levels.levels|No entry point for level found: /levels/garage. Ignoring level.                      055.613|E|libbeamng.lua.V.ResearchCom|Error reading from socket: closed                                                 057.143|I|libbeamng.lua.V.ResearchVE|Trying to connect to: localhost:64257                                              059.355|I|libbeamng.lua.V.ResearchVE|Could not connect...                                                               060.001|I|GELua.util_researchGE.ResearchGE|Trying to connect to: localhost:64256                                        062.317|I|GELua.util_researchGE.ResearchGE|Could not connect...                                                         063.386|I|libbeamng.lua.V.ResearchVE|Trying to connect to: localhost:64257                                              065.533|I|libbeamng.lua.V.ResearchVE|Could not connect...                                                               065.735|I|GELua.util_researchGE.ResearchGE|Trying to connect to: localhost:64256                                        067.760|I|GELua.util_researchGE.ResearchGE|Could not connect...                                                         074.911|I|GELua.util_researchGE.ResearchGE|Trying to connect to: localhost:64256 `

What could be the problem?

alessiogambi, 2019-08-19:
Hi,

"https://projects.beamng.com/projects/research/files is empty." this is always been the case. The correct files are under: https://projects.beamng.com/projects/research/repository/show/trunk

Regarding "055.442|E|GELua.ResearchCom|Error reading from socket: closed", it's a wild guess but did you check if you have firewall rules which prevent opening sockets from python?

autonomobil, 2019-08-20:
Hey, thanks for the answer

At the bottom of the [beamng research homepage](https://beamng.gmbh/research/ ) under DOWNLOAD it has a link to `https://projects.beamng.com/projects/research/files`  ...so that should maybe be changed

I will check my firewall

FitashUlHaq, 2019-10-22:
Hi, I am facing the same problem. Can you please help me solve it?

FitashUlHaq, 2019-10-22:
In my case it connects for the first time and then cannot connect

029.612|I|libbeamng.lua.V.ResearchVE|Trying to connect to: localhost:64257
031.627|I|libbeamng.lua.V.ResearchVE|Connected!
031.630|I|GELua.util_researchGE.ResearchGE|Setting engine flags!
031.662|D|GELua.scenario_scenarios.scenarios|changeState: running
031.663|D|GELua.core_camera.|Camera switched to "orbit"
031.712|E|GELua.core_levels.levels|info.json missing: /levels/garage/info.json
031.713|E|GELua.core_levels.levels|No entry point for level found: /levels/garage. Ignoring level.
031.714|E|GELua.core_levels.levels|info.json missing: /levels/GridMap/info.json
031.715|E|GELua.core_levels.levels|No entry point for level found: /levels/GridMap. Ignoring level.
031.819|E|GELua.ResearchCom|Error reading from socket: closed
031.874|E|libbeamng.lua.V.ResearchCom|Error reading from socket: closed

oooooverflow, 2019-10-22:
yeah,my problem is exactly the same as yours.

043.818|I|libbeamng.lua.V.ResearchVE|Trying to connect to: localhost:64257
045.843|I|libbeamng.lua.V.ResearchVE|Connected!
045.847|I|GELua.util_researchGE.ResearchGE|Setting engine flags!
045.879|D|GELua.scenario_scenarios.scenarios|changeState: running
045.880|D|GELua.core_camera.|Camera switched to "orbit"
045.940|E|GELua.core_levels.levels|info.json missing: /levels/garage/info.json
045.942|E|GELua.core_levels.levels|No entry point for level found: /levels/garage. Ignoring level.
045.944|E|GELua.core_levels.levels|info.json missing: /levels/smallgrid/info.json
045.947|E|GELua.core_levels.levels|No entry point for level found: /levels/smallgrid. Ignoring level.
045.974|E|GELua.ResearchCom|Error reading from socket: closed
045.981|E|libbeamng.lua.V.ResearchCom|Error reading from socket: closed
048.730|I|libbeamng.lua.V.ResearchVE|Trying to connect to: localhost:64257
052.758|I|libbeamng.lua.V.ResearchVE|Could not connect...

FitashUlHaq, 2019-10-22:
I tried
1- turning off firewalls
2- running it as administrator
3- freeing all ports
None of above worked

FitashUlHaq, 2019-10-22:
Hi @oooooverflow and @autonomobil, I solved the problem using the windows powershell. it does not work if you run it from cmd. 

If you write the commands on powershell it works perfectly fine!

vladimirwest, 2019-10-24:
The game is running perfectly when python script is executing.
So simplest solution that worked for me is add  `time.sleep(99999)` to the end of the main python script.

manojhuilgol, 2019-11-15:
@vladimirwest Can you please describe where exactly did you add ```time.sleep()``` ?
I tried adding it inside ```def main()``` function, but still it is not working.

closed 2021-03-05 by Palculator