BeamNG/BeamNGpy #30: Export data to CSV table

vladimirwest, 2019-09-13:
Hi there!
I found that BeamNG could fit well for my research. Now I'm looking for a way to collect data from the game.
In the youtube demonstration from Engineering psychology I found a great example of what i need (1:30-1:35):
[image]

I want to export at least:
> position
> speed
> wheel angle
> collision between cars (if the car is damaged)

Is there a simple option to export such data to csv?

alessiogambi, 2019-09-13:
Hi, yes this should be simple enough.

Note that there's a bunch of students doing research on BeamNG.research, and they might have already implemented what you are looking for. 

I can add you to our slack channel Driversity.slack.com if you like, so you can ask them yourself and send a pull request with the solution here.

vladimirwest, 2019-09-13:
Sure.
Please send it to `vvkilyazov@edu.hse.ru`, thanks!

closed 2020-09-07 by pascale-commits