BeamNG/BeamNGpy #31: get vehicle bounding box

qazi2, 2019-10-03:
Hi guys,

I am not able to get vehicle bounding box by given `bng.get_vehicle_bbox(vehicle2)` after starting the scenario.
```bng = beamng.open()
bng.load_scenario(scenario)
bng.start_scenario()
vehicle.ai_set_line(path1)
vehicle2.ai_set_line(path2)
vehicle3.ai_set_line(path3)
bng.get_vehicle_bbox(vehicle2)
# Following error occurs:

bbox = bng.get_vehicle_bbox(vehicle2)
  File "D:\Beamng_qazi_data\Thesis_final_code\venv\lib\site-packages\beamngpy\beamng.py", line 1130, in get_vehicle_bbox
    resp = self.recv()
  File "D:\Beamng_qazi_data\Thesis_final_code\venv\lib\site-packages\beamngpy\beamng.py", line 241, in recv
    return recv_msg(self.skt)
  File "D:\Beamng_qazi_data\Thesis_final_code\venv\lib\site-packages\beamngpy\beamngcommon.py", line 155, in recv_msg
    length = int(str(length, 'ascii'))
```

According to the function it should return: 
near_bottom_left: Bottom left point of the near rectangle as an (x, y ,z) triplet
• near_bottom_right: Bottom right point of the near rectangle as an (x, y, z) triplet
• near_top_left: Top left point of the near rectangle as an (x, y, z) triplet
• near_top_right: Top right point of the near rectangle as an (x, y, z) triplet
• far_bottom_left: Bottom left point of the far rectangle as an (x, y, z) triplet
• far_bottom_right: Bottom right point of the far rectangle as an (x, y, z) triplet
• far_top_left: Top left point of the far rectangle as an (x, y, z) triplet
• far_top_right: Top right point of the far rectangle as an (x, y, z) triplet

Palculator, 2020-05-14:
I can't reproduce this. I've created a small unit test that spawns two vehicles, makes one drive across the map, and compares the bounding box of the moving vehicle before driving and after 2000 frames to ensure the bounding box moves: https://github.com/BeamNG/BeamNGpy/blob/master/tests/test_vehicle.py#L102

No exceptions when running it and the assertions hold true.

alessiogambi, 2020-05-14:
@Palculator: @qazi2 was one of my students in Passau working on creating simulations from videos. He might have misunderstood the meaning of BBox in this context. So to clarify, BBox given by BeamNG.py are NOT the bounding boxes in the image around the cars.

BTW, this might be a very nice feature to add to the camera sensor!

In summary, I think you can close this issues with no remorse.

Palculator, 2020-05-14:
Yeah, object bounding boxes for the camera sensor are on my to-do list.

closed 2020-05-14 by Palculator