BeamNG/BeamNGpy #38: get projective 3D point  to camera pixel coordinate

joshuafc, 2019-12-18:
have a 3d point (x, y ,z) from Vehicle.get_bbox() .
how can I get the associated point at camera pixel coord.
Or how can I get the camera model and parameters, I can do the compute myself.

pascale-commits, 2021-01-08:
Thanks for the input. It might be too late for your project but I've opened up this feature request for discussion in the team.

BleedObsidian, 2021-03-16:
I also vouch for this feature! Being able to map camera space to top-down world coordinates is a very common requirement when using cameras attached to a vehicle in real life.

dstark481, 2022-08-22:
This feature will be available in the upcoming release.

dstark481, 2022-09-23:
FYI

This feature is now available in our most-recent release.  We have a member function `convertWorldPointToPixel()` which can be called on a camera sensor to convert a given world-space point (x, y, z) to the pixel (x, y) on the camera.

Full documentation on the new camera sensor can be found here:
https://documentation.beamng.com/beamng_tech/sensors/camera/

Best regards,
BeamNG Tech Team

BleedObsidian, 2022-09-24:
Amazing work guys, thank you, this will soon be the best AV simulation environment out there!

closed 2022-09-30 by joshuafc