BeamNG/BeamNGpy #41: Set AI Path should not teleport the vehicle

alessiogambi, 2020-04-17:
In some situations calling the following code:
```self.vehicle.ai_set_mode('disabled')
self.vehicle.ai_set_script(self.script)```
results in teleporting the car, which is somehow an unexpected behavior. This might be something related to the "current speed" of the car which is zero at the beginning or it might be something else.

alessiogambi, 2020-04-17:
I see inside script.ai that there's a variable named `loopType` which decides whether or not to teleport the car. However, I cannot understand why my script triggers such behavior. My Ai Script is defined starting from the current position of the car but somehow the car is moved back by some distance? The only idea I have is that the car cannot reach the point in time for some reason so it needs more space to drive? But in this case, why not driving slower?

alessiogambi, 2020-04-17:
I can explain this behavior as follow: in my code, I do not specify the initial dir and the up dir so BeamNGpy infers them from the first two nodes of the script, which happen to be placed on a turn in my case. Because of this, the computed direction is different than the current one, and BeamNG teleports my car accordingly (to fit both the given direction and "distance" to the first node)

Palculator, 2020-06-03:
This should be fixed in v1.16. The vehicle will now never be teleported and will simply drive to the first point in the script. If you want to teleport it nonetheless, it's possible with: https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.beamng.BeamNGpy.teleport_vehicle

closed 2020-06-03 by Palculator