BeamNG/BeamNGpy #42: Loading and staring existing scenarios does not work

alessiogambi, 2020-04-18:
I am trying to load and start existing scenarios so I do:
- Start beamng:
    beamng = BeamNGpy('localhost', 64256)
    bng = beamng.open(launch=True)

- Define and (find) the existing scenario: 
    scenario1.bng = bng
    scenario1.find(bng)

- Load the scenario:
    bng.load_scenario(scenario1)

- Start the scenario:
    bng.start_scenario()

However, after calling load scenario (and I can see that the scenario is loaded), start scenario blocks.

alessiogambi, 2020-04-18:
A possible workaround is to introduce a sleep between load_scenario and start_scenario. However, a better solution would be to return from load_scenario at a point in which start_scenario can be actually called or expose a method that enables checking whenever is safe to start the scenario.

Palculator, 2020-06-03:
Should be fixed in v1.16. I found a better way to check if a scenario has finished loading (nothing ugly like sleeping 😜)

closed 2020-06-03 by Palculator