BeamNG/BeamNGpy #46: MultiShot Camera

alessiogambi, 2020-05-15:
When collecting data to train AI, one way to improve the data set is to collect images from a slightly different position of the camera and slightly different rotations. This might be achieved by declaring a number of Camera components, or maybe by pausing the simulation and taking all the screenshots with the same camera sensor (but under different configurations), however, a nice meta-sensor/sensor that gets a list of configuration might be easier to use? Such a sensor might be also configured to generate random pose/rotations of the camera while the camera points towards the same point?

Palculator, 2020-06-03:
There is no inherent limit to the amount of cameras you can add to a vehicle. You could simply add `n` cameras at `n` different positions/directions. I looked into implementing a camera that has multiple positions/directions itself, but it would mainly clutter up the code to save having to create multiple `Camera` objects instead of just multiple positions/directions. With the codebase as it is right now, it's also kind of a hassle to have a sensor cause multiple polls in a single call. This can be changed, of course, but would require more work and in the meantime simply using multiple `Camera` objects is probably easier. I'm open to other suggestions, though.

alessiogambi, 2020-06-03:
Well, if the properties of the camera can be changed dynamically, one can attach a single camera sensor, pause the simulation, collect all the required images, and resume the execution.

Palculator, 2020-06-04:
Yup, this was already possible. I've added an example here: https://github.com/BeamNG/BeamNGpy/blob/master/examples/multishot_camera.ipynb

Make sure to update BeamNGpy, though, since writing the example actually made me find & fix an issue with camera orientation.

closed 2020-06-04 by Palculator