BeamNG/BeamNGpy #48: BeamNG should allow for multiple connections

alessiogambi, 2020-05-22:
At the moment, only one BeamNGpy instance can be used to connect to a simulation. It would be ideal if, at least for vehicles, one can use multiple sockets/end-points, such that one BeamNG instance might be used to control the scenario while the others control the vehicles?

Palculator, 2020-06-03:
You are right, this would be neat. However the code is mostly written with the Python side acting as the server, BeamNG.drive as the client. In hindsight this was a mistake I made early on, but it's one I have to deal with now so it could take a bit to rework everything to allow multiple concurrent clients.

Palculator, 2021-03-05:
Added in v1.19.1: https://github.com/BeamNG/BeamNGpy/blob/master/CHANGELOG.rst
Example here: https://github.com/BeamNG/BeamNGpy/blob/master/examples/multi_client.ipynb

closed 2021-03-05 by Palculator