BeamNG/BeamNGpy #49: Quey interface

alessiogambi, 2020-05-22:
Would it be possible to query a running simulation from BeamNGpy? For example, getting how many (and which) cars are running, etc.

Palculator, 2020-06-03:
Good idea. There already exists a function with which you can query for objects in the environment: https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.beamng.BeamNGpy.find_objects_class

However, this probably requires some more documentation as the class of objects are specific to the game (and don't correspond to, for example, the `Vehicle` class.) What's missing is a way to query active vehicles and I will probably add one for that — ideally one that returns vehicle objects you can attach sensors to and control like all others.

Palculator, 2021-03-05:
Added in v1.19.1: https://github.com/BeamNG/BeamNGpy/blob/master/CHANGELOG.rst
Most notably the `get_current_vehicles` and `sync_scene` methods of `BeamNGpy` and `Scenario` classes. This will  be expanded upon in the future but for now should enable a bunch of things already.

closed 2021-03-05 by Palculator