BeamNG/BeamNGpy #50: Customize Car Attributes

alessiogambi, 2020-06-17:
Is there a way to customize the attributes of a car? 
Like changing its color, tyres, whatever?

alessiogambi, 2020-09-03:
Hi, thanks for the example.

> As far as I can see the python API does not support the configuration of any other vehicle properties but it is possible to change them in-game, though I guess this is not what you're interested in.

Actually, this is exactly what I am interested in. Maybe I should open a different issue or refine the description of this one?

pascale-commits, 2020-09-03:
Please add a more detailed description of what you are searching for :) I've reopened the issue.

pascale-commits, 2020-09-07:
Thanks for specifying. 

It is possible to change main color of a car on the fly in the game. This is supported by the beamngpy lib through the <set_color> method of the vehicle class.
Changing other colors on the fly is not supported at the moment, but it is possible to change the available vehicle colors manually in the game in a given level by pressing ESC -> Selecting Vehicle Config -> Select Color Tab and adjusting the colors to your needs.

Setting new skins triggers a vehicle reload, so it is not possible to change them on the fly.

Yes, it is not possible to change any part of the vehicle that affects the deformation mesh. It is not possible to transfer the damage/deformation of  one beam structure to another.

alessiogambi, 2020-09-07:
Great. Thanks for the clarification!

closed 2020-09-07 by pascale-commits