BeamNG/BeamNGpy #51: Import new cars models, objects, elements in the scenarios?

alessiogambi, 2020-06-17:
How one could import new elements/models into a scenario?
Should those elements be already available during "scenario generation time" or would it possible to import them on the fly from "file" or "URL"?

pascale-commits, 2020-08-25:
There are two options to generate assets through the BeamNG library BEFORE scenario generation:

1) Using the various available procedural objects (ProceduralCylinder, ProceduralCone, ProceduralCube) for which you can find an example here: [examples/procedural_meshes.py
](https://github.com/BeamNG/BeamNGpy/blob/master/examples/procedural_meshes.py)
2) Using the ScenarioObject class to spawn any asset available in the game. Here a minimal example:

`from shapely.geometry import MultiLineString
from matplotlib import pyplot as plt

from beamngpy import BeamNGpy, Vehicle, Scenario, ScenarioObject

beamng = BeamNGpy('localhost', 64256)

scenario = Scenario('east_coast_usa', 'spawn_objects_example')

vehicle_pos = (647.596008,-145.919678,53.4411545)
vehicle_rot = (0.00938976184,-0.00648232969,0.325634241)
vehicle = Vehicle('ego_vehicle', model='pickup', licence='PYTHON')
scenario.add_vehicle(vehicle, pos=vehicle_pos, rot=vehicle_rot)

rb_pos = (640.05011,-156.296127,53.5653763)
rot = (0,0,0)
scenario_obj = ScenarioObject(oid='roadblock', 
                              name='sawhorse',
                              otype='BeamNGVehicle',
                              pos=rb_pos,
                              rot=rot,
                              scale=(1, 1, 1),
                              JBeam = 'sawhorse',
                              datablock="default_vehicle"
                              )
scenario.add_object(scenario_obj)

scenario.make(beamng)

bng = beamng.open()
bng.load_scenario(scenario)
bng.start_scenario()
`

Concerning the generation of assets AFTER scenario creation, i.e. in-game:
It is possible to spawn assets of type BeamNGVehicle during the simulation with the spawn_vehicle and despawn_vehicle function  of the BeamNGpy class. I'll write an example for that as well in the near future :-)

Hope that helps!

pascale-commits, 2020-09-03:
I wrote a small example, which includes spawning after scenario creation, though it is not available on the dev branch yet, but you can find it here: [examples/spawning.ipynb](https://github.com/BeamNG/BeamNGpy/blob/quaternions/examples/spawning.ipynb). You will have to change the rotation specification for it to work in your beamngpy version right now, or you wait for it to be integrated into dev/master :)

closed 2020-09-03 by pascale-commits