BeamNG/BeamNGpy #54: BeamngPy refuses connecting to beamng research

Firestorm7893, 2020-06-26:
Hello, I have been trying for a while, but i just can't get BeamngPy to connect. Beamng Research opens, but on the python console it looks like it's stuck. After a very long time i get a "socket timed out" error and nothing else.
I tried running as administrator, allowing everything trough the firewall, but it just won't connect :(

(i'm using the vehicle state example)

Any ideas to why this might be happening?

Thanks in advance.

alessiogambi, 2020-06-26:
Hi, 
are you loading an existing scenario or are you creating a new one from scratch? Can you check that no other instances of BeamNG.research are running somehow?

Firestorm7893, 2020-06-26:
> 
> 
> Hi,
> are you loading an existing scenario or are you creating a new one from scratch? Can you check that no other instances of BeamNG.research are running somehow?

The example creates a new scenario, which i can select manually from Beamng.
 I tried rebooting the computer but still nothing, I get Timeout :(

BleedObsidian, 2020-06-29:
@Firestorm7893, have a look at this issue I created and see if it solves your problem https://github.com/BeamNG/BeamNGpy/issues/55

Firestorm7893, 2020-06-30:
@BleedObsidian that fixed it thanks a lot

closed 2020-06-30 by Firestorm7893