BeamNG/BeamNGpy #55: BeamNG.Research Not Connecting - module 'util/trackBuilder/proceduralPrimitives' not found.

BleedObsidian, 2020-06-29:
The python server from BeamNGPy is never receiving a connection from BeamNG.research.

I believe the reason is due to the following error preventing BeamNG.research from getting to the point where it attempts to connect to the python server. 

[image]

I've checked and indeed, the mentioned module seems to be missing:

[image]

I've also checked multiple revisions of the SVN repository and it's also not there.

Any ideas why this is happening? Im pretty certain this error is why im not receiving a connection, because having looked at the the researchGE.lua file, this is clearly where the connection is made so I imagine this error is causing it to fall over.

BleedObsidian, 2020-06-29:
Managed to find a workaround. The missing proceduralPrimitives.lua file was present in the full BeamNG.drive game available from steam. I managed to copy this file from their into the relevant place for the research version.

It would appear that the file is simply missing from the SVN repository?

Heres the file for anyone that also has this issues and does not want to buy the steam game: https://drive.google.com/file/d/14SzDOndAN7FB5LB8dhLJJb7VLS_3Q7Ix/view?usp=sharing

Firestorm7893, 2020-07-02:
Many things look like they are out of date, for example in [this example](https://github.com/BeamNG/BeamNGpy/blob/master/examples/vehicle_state_plotting.ipynb) the section that says (or any section really) `display(sensors['electrics']['values']['wheelspeed'])` will not work, since the **['values']** keyword is not accepted anymore and must be removed. 
There are things that aren't really explained in the documentation
For example, how to use another veichle other than the etk800. 
I tried importing a sunburst from the main game and changing the "model" in the vehicle constructor to sunburst, but no luck.

BleedObsidian, 2020-07-02:
I've also noticed that many of the examples use the GridMap, but the latest version from the SVN repository does not include the GridMap, only "west_coast_usa" ([example](https://github.com/BeamNG/BeamNGpy/blob/master/examples/road_definition.py))

I've tried to include the GridMap from the main BeamNG.drive game but there are all sorts of material linking errors that occur that make the map not render probably (but successfully load).

alessiogambi, 2020-07-02:
Hi,
Are you using by chance the beamng research version that is linked to the web site? The one that you can access without being registered/enabled? In that case you should know that that version is outdated and more importantly strongly limited. For example, you can use only one car and one model. Additionally that version is not compatible with beamngpy. If you are interested in using the updated/unlimited version of the simulator you need to get access by sending an email to beamng. There might be conditions to get it. For example you must be an active student, use the simulator for research purposes or personal use and the like.

Palculator, 2020-07-03:
Yeah, as Alessio pointed out, that version is no longer maintained. We should make that more clear and there will be a replacement soon. In the meantime you can contact us at research@beamng.gmbh to sort out if you're entitled to access of the advanced version.

closed 2020-07-03 by Palculator