BeamNG/BeamNGpy #56: BNGValueError Scenario Not Found - But the game proceeds to load the scenario fine.

BleedObsidian, 2020-06-29:
Once the game connects to the python server and the load scenario call is made. The game returns a BNGValueError about not being able to find the scenario (this then means the python can not continue) but the game loads the scenario fine?

[image]

[image]

Any ideas why this is happening? Im currently trying to dig through the lua to prevent the failed to load error being sent to the python server.

BleedObsidian, 2020-06-30:
I'm stumped on this one, I had a wild attempt of placing the BeamNGPy.zip mod file in the mods folder for the full steam BeamNG game and I managed to get much further. The scenario is recognised as found and the python server is told so successfully.

However everything works except from the research specific features such as LiDar and I imagine many other things such as cameras:

[image]

I assume this is why there is a research specific BeamNG version. Everything is leading me to believe that the svn repository for BeamNG.research is wildly out of date or in a corrupt state of broken?

@Palculator

alessiogambi, 2020-06-30:
@BleedObsidian what version of beamng research are you using? Does beamngpy warn you about a version mismatch between the api and the simulator?

Note that despite being base on the same technology the game and the simulator are different beasts. I believe it might be hard to make the game behave as the research version and the other way around.

BleedObsidian, 2020-06-30:
Hi @alessiogambi, thanks for the reply.

I'm using the latest version of BeamNG.research (revision 15 of the svn repository) and this is the version information from the console: `v 1.4.0.0.0 - x64 - build 0 - 6/30/2020`

For the version of BeamNGPy, i'm using the latest available from the python pip3 repository, 1.16.3. Which I'm now realising might be out of date, let me trying using the latest BeamNGPy from source.

In terms of getting a version mismatch error, I do not.

BleedObsidian, 2020-06-30:
@alessiogambi 

I've just tried installing the pip package from source and also from the dev branch. Same issue.

It doesn't make much sense to me, I can confirm that both BeamNGPy and BeamNG.research are using the correct user path.

I've also tried on a brand new machine from scratch with a fresh install of BeamNG.reserach and BeamNGPy, im getting the same issue? I'm sure this issue can't just be me.

Firestorm7893, 2020-06-30:
I'm having the exact same issue, i even tried going into beamng.py and hardcoding the path to the scenario but nothing works

BleedObsidian, 2020-06-30:
@Firestorm7893 

Really glad this isn't just me!

It's been driving me nuts to try and figure this out. I'm really keen to use this as a simulator for my dissertation next year but I just can't get it to work.

BleedObsidian, 2020-07-01:
@Firestorm7893 @alessiogambi @Palculator

I've found a fix for the issue. I have no clue how anyone has been using BeamNG.research recently, this bug should be affecting everybody.

In ResearchGE.lua:

[image]

If the value of `scenariosLoader.startByPath()` returns nil, then the scenario is assumed to have not been found. However, the value of  `scenariosLoader.startByPath()`  **always returns nil**...

In lua/ge/extensions/scenario/ScenariosLoader.lua:

[image]

In my console logs posted earlier, you can see that the message `log('E', logTag, path .." does not exist")` did not appear in the console. This told me that `FS:fileExists(path)` was clearly passing and the game was successfully loading the scenario.

Adding a `return true` fixed the issue and no longer falsely sends a BNGValueError to the python server and everything is now working perfectly:

[image]

The only problem is, only BeamNGPy appears to be open source and this file is clearly part of the BeamNG core lua API. This means I can't make a pull request to fix the issue without moving the file checks to the ResearchGE.lua file which isn't a very clean solution. Will the devs please update and include this fix in ScenarioLoader.lua?

BleedObsidian, 2020-07-01:
I've just checked the ScenarioLoader.lua for the most recently copy of the steam version of BeamNG.drive and the return true is already there! This clearly is showing that the SVN repository is out of date or broken along with the missing proceduralPrimitives.lua file: https://github.com/BeamNG/BeamNGpy/issues/55

BleedObsidian, 2020-07-11:
Closing this issue as it is only a problem with the publicly available SVN repository version, a fully working version is available when contacting: research@beamng.gmbh

closed 2020-07-11 by BleedObsidian