BeamNG/BeamNGpy #57: Run custom scenario from Beamngpy

vladimirwest, 2020-06-30:
Hello!

I have created a scenario with some waypoints from the level editor.

Is it possible to run that scenario from beamngpy?

masskro0, 2020-06-30:
Hello,

you can select your own scenario like that:
`scenario = Scenario('levels_folder', 'example_scenario')
scenario.path = Path(join(ENV["BNG_HOME"], "levels", "levels_folder", "scenarios"))`
E.g. if you made a custom west_coast_usa scenario, you can replace "levels_folder" with "west_coast_usa" and "example_scenario" with the name of your scenario (without extensions). Instead of joining folders and using the BNG_HOME environment variable, you can also give the full path to your scenario as a string.

vladimirwest, 2020-06-30:
> Hello,
>
> you can select your own scenario like that:
> `scenario = Scenario('levels_folder', 'example_scenario')`
> `scenario.path = Path(join(ENV["BNG_HOME"], "levels", "levels_folder", "scenarios"))` [...]

Thanks, this worked perfectly.
Closing the issue.

closed 2020-06-30 by vladimirwest