BeamNG/BeamNGpy #59: Update object position

masskro0, 2020-07-05:
I need to update the position of static objects while the simulator is running but I can't find a way to do this. There is a teleport vehicle function and I tried to do the same for objects but it didn't work. Is there a way to update object positions?

Background: I need to control traffic lights but the wca level uses a png file to change the light of traffic light colors, which can't be controlled dynamically by the code. I also couldn't find a way to add lights (e.g. PointLight) only using beamngpy. That's why I thought about adding a red, yellow and green ProceduralRing object which should jump to the right position.

closed 2020-07-20 by masskro0

masskro0, 2020-07-20:
I found a way to change the position of any object by using Lua. E.g.:
`local light_0 = scenetree.findObject("traffic_blinking_0")`
`local p1_0 = Point3F(85.12596722509497, -87.33277457707955, -33)`
`light_0:setPosition(p1_0)`

alessiogambi, 2020-07-20:
@masskro0 Hi, this is really a good catch. Maybe you can create a merge request on BeamNGpy to enable moving objects around during the simulation from python? 

I wonder: will one be able to add static objects to the simulation **dynamically**?