BeamNG/BeamNGpy #60: BeamNG loads but no ai works

ASionville, 2020-07-08:
Hello !
I have a problem : I installed BeamNGpy and BeamNG.research, my folder looks like that :
[image]

and in the main.py I pasted the code for west_coast with lidar (found in /examples) and changed only this line to add the path ;
[image]

The game loads to the main menu and then, I must go to scenarios and load the right one (I looked in the code and I think it's not supposed to do that). Once the scenarios loaded and I started it, nothing happens and I have to drive the car myself.

Here a file with the BeamNG console output just after I start the scenario ( after the 3 2 1 go) :
[output.txt](https://github.com/BeamNG/BeamNGpy/files/4893628/output.txt)

Can you please help me !
ASionville

alessiogambi, 2020-07-09:
Hi, are you using the unlimited version or the basic version linked to the
beamng web site? The one from the web site is likely outdated.

masskro0, 2020-07-09:
You can also try to set the BNG_HOME environment variable or give a full path as a string instead of just "beamng_research/trunk" (also note that on Windows you need two "\\" to separate folders). Do you get an exception in your IDE? If the scenario doesn't load automatically then the code execution will stop before it gets to the AI part.

ASionville, 2020-07-09:
@alessiogambi I am using the SVN version available on the website but they will unlock me the access for the unlimited version soon. I will try again with it.

@masskro0 If the relative path doesn't work (although it seems too as it starts the game), I will try the absolute one. (Thanks for the advice about Windows paths)

ASionville, 2020-07-14:
@alessiogambi was definetely right, the "limited" versionj seems to be broken, now it works perfectly !

closed 2020-07-14 by ASionville