BeamNG/BeamNGpy #61: Beamng.research release clarification

alessiogambi, 2020-07-09:
Since this seems to be a recurrent issue, @Palculator maybe can clarify somewhere what is the required version/release of the beamng.research simulator that beamngpy needs?

A note on the Readme may be sufficient.

pascale-commits, 2021-01-08:
Thanks for the input, we'll address this issue soonish.

Palculator, 2021-03-05:
Both of them are kind of developed hand-in-hand and the latest version of BeamNGpy should always work with the latest version of BeamNG.tech. I think if research depends on specific versions of either it should note that as such. Some confusion can arise from people trying to use BeamNGpy with BeamNG.drive, which might work to an extent, but is not officially supported and I've added a note about that to the README.

closed 2021-03-05 by Palculator