BeamNG/BeamNGpy #62: BNG.Research multi-threading

NismoR, 2020-07-10:
Hello, I'm trying to separate the vehicle control from the sensor polling on separate threads. Is the BNG.Research capable of that at all? 
Here is my code what I'm trying to run:
[https://pastebin.com/VegQmHSU](url)

And here are the error messages that I've got:
[https://pastebin.com/TN9KXYnx](url)

alessiogambi, 2020-07-11:
Hi
I cannot access your code at the moment so I cannot comment on it. But AFAIK beamngpy assumes that only one instance is connected at the time, and instances are not thread safe. that is handling of messages/interactions with the simulator must be sequential and your code has to handle sharing access to the API.

NismoR, 2020-07-13:
Yes, it was my fault, I've edited the comment so now the code is accessible.
I'm sad to hear that, but thank you for your quick reply.
I hope they'll change it, since the unknown delays can cause disturbances in a faster control loop.

alessiogambi, 2020-07-13:
@NismoR: Do your control loops work in _real_-time? Otherwise, the solution is to run the simulation synchronously. You can pause the simulation, read and process all the data, send control commands back, and resume the simulation exactly from where it was paused. There should be also a sensor that gives you the "simulation time," in case you need it.

pascale-commits, 2020-10-28:
Multi-threading is not a planned feature. Use the solution suggested by @alessiogambi for now.

closed 2020-10-28 by pascale-commits