BeamNG/BeamNGpy #63: Memory Error when using Lidar

ASionville, 2020-07-14:
Hi there !

I am having an issue using the lidar :

I can run the west_coast_lidar.py example but the code down below doesn't work
This window appears everytime : 
[image]

Can you please tell me what am I doing wrong ? I have 24Go of RAM so it feels strange that it's not enough.

```python
import sys

from time import sleep

import numpy as np

from beamngpy import BeamNGpy, Scenario, Vehicle, setup_logging
from beamngpy.sensors import Lidar

def main():
    setup_logging()

    beamng = BeamNGpy('localhost', 64256, home="../beamng-research_unlimited/trunk")

    # Create a scenario in west_coast_usa
    scenario = Scenario('west_coast_usa', 'research_test',
                        description='Random driving for research')

    # Set up first vehicle, with two cameras, gforces sensor, lidar, electrical
    # sensors, and damage sensors
    vehicle = Vehicle('ego_vehicle', model='etk800',
                      licence='AUBIN IA', color='White')

    lidar = Lidar(offset=(0, 0, 1.6), visualized=False)

    # Attach them
    vehicle.attach_sensor('lidar', lidar)

    scenario.add_vehicle(vehicle, pos=(-717.121, 101, 118.675), rot=(0, 0, 45))

    # Compile the scenario and place it in BeamNG's map folder
    scenario.make(beamng)

    # Start BeamNG and enter the main loop
    bng = beamng.open(launch=True)
    try:
        bng.set_steps_per_second(60)  # With 60hz temporal resolution

        # Load and start the scenario
        bng.load_scenario(scenario)
        bng.start_scenario()
        # Put simulator in pause awaiting further inputs
        bng.pause()

        assert vehicle.skt
        # Send random inputs to vehice and advance the simulation 20 steps
        while True:

            bng.step(60)

            # Retrieve sensor data and show the camera data.

            sensors = bng.poll_sensors(vehicle)

            print(sensors['lidar'])
            sleep(1)
    finally:
        bng.close()

if __name__ == '__main__':
    main()```

Palculator, 2020-07-16:
Hello, unfortunately, neither my colleague or I seem to be able to recreate this issue, but we have seen it before in other users. The only real recommendation I can give is making sure your graphics drivers are fully up to date or doing a clean install of them, maybe reinstalling/upgrading DirectX and so on. If you google that specific error code with "beamng" you'll find various people randomly having it and those things solving it: https://www.beamng.com/threads/d3d11-error-8007000e.54495/

mihalykorodi, 2021-02-11:
Hi All!

I am facing this issue too. It happens exactly on the second sensor poll, and driver update, cache clearing did not help.

davidp-ro, 2021-04-07:
@mihalykorodi Hi there, have you managed to find a fix/workaround for this? I have basically the exact same issue (on an RX590, and unfortunately I don't have an Nvidia Card on hand) and nothing seems to help (downgrading drivers, updating drivers, reinstalling, all to no avail)...

closed 2026-01-05 by EPI-BeamNG