BeamNG/BeamNGpy #64: Change Environmental Properties

alessiogambi, 2020-07-15:
It would be cool/nice if BeamNGpy let us change the time-of-day, the sky, and the weather (or any other environmental properties that can be configured throughout the world editor)

masskro0, 2020-07-16:
In beamng.py there is a function called "set_weather_preset" where you can load a weather preset from art/weather/default.json. You can add your own presets as well. This function also allows to change the time of day. Otherwise you can change it via the function "set_tod(tod)".

alessiogambi, 2020-07-20:
I see. Should the user ensure that some definition of weather (e.g., heavy rain, light rain) are provided **before** the simulation is started or the various weather conditions will be always available and can be changed on-the-fyl?

masskro0, 2020-07-20:
You can call this function right after bng.load_scenario. Also, you can change it on-the-fly but it takes a few seconds to load the weather preset.

pascale-commits, 2021-01-07:
Closing, since this issue seems to be solved. Thanks for you input.

closed 2021-01-07 by pascale-commits