BeamNG/BeamNGpy #66: Change color of same static object

masskro0, 2020-07-22:
Hello,
let's say I place 10 times the same object, is it possible to give each one a different color?  One method would be to duplicate the object multiple times but this is very space consuming.

pascale-commits, 2021-01-08:
Hi! Can you specify the use case?

For objects that would be part of the prefab, you could make use of the `scenario.add_object` method. You can find an example of it [here](https://github.com/BeamNG/BeamNGpy/blob/master/examples/spawning.ipynb). Adding `color = "0.3 0.30.3"` to the constructor call should do the trick.

Available procedural objects do not support the color option.

masskro0, 2021-01-10:
Hello,
I'm using a car model in a single DAE format to add static parked cars without physics. Right now, I'm creating a StaticObject like this:
```
self.scenario = Scenario('urban', 'urban_{}'.format(self.index))
...
golf = StaticObject(pos=pos, rot=rot, name=name_car,
                                    scale=(1.2, 1.2, 1.2), shape='/vehicles/87Golf/87Golf.dae')
self.scenario.add_object(golf)
```
Right now, I'm changing the color manually in the materials.cs file, which allows only one color for all parked cars for each simulation run. The color is not set in the prefab file.