BeamNG/BeamNGpy #67: "running" value of electrics sensor always returns True

masskro0, 2020-07-29:
If I check the value "running" of the electrics sensor then it always returns True, even if I turn off the engine. I'm using beamngpy 1.15 with BeamNG 1.9.

pascale-commits, 2020-09-08:
You're right, I can't resolve this problem at the moment, but if you need this functionality I can give you  sort of a hot fix which you can copy and paste into your local beamngpy installation.

In your virtual environment (if you've used `python -m venv...`, but anaconda should have a similar setup), you can find researchVE.lua in the following path: <venv root dir>/Lib/site-packages/beamngpy/lua/researchVE.lua

In line 168 (if you are up to date with master) should be this code snippet:
`sensorHandlers.Electrics = function(msg)
  local resp = {type = 'Electrics'}
  resp['values'] = electrics.values
  return resp
end`

which you can replace with this:
`sensorHandlers.Electrics = function(msg)
  local resp = {type = 'Electrics'}
  resp['values'] = electrics.values
  resp['values']['running'] = controller.mainController.engineInfo[18]
  return resp
end`

I'll get back to this issue in 2-3 weeks with a better solution.

masskro0, 2020-09-08:
Thank you! Your solution worked for me. Since I'm using an older version of BeamNGpy, the file was located in trunk/lua/vehicle/extensions.

closed 2020-09-08 by masskro0

pascale-commits, 2020-10-08:
Closing remark on this issue from our side: 
The values `ignition` and `running` won't work with the current BeamNG.research version but this issue should be fixed for any future BeamNG.research version > 1.7.0.0 

Until then you can use the hack I wrote down above, though that might require a clean install for any BeamNGpy updates. 
For the `ignition` value there is no easy hack, so I won't write it down here. Anyone really needing that value can comment on this thread and we'll see what a good solution would be.