BeamNG/BeamNGpy #68: Query roads properties

alessiogambi, 2020-08-05:
BeamNGpy allows to get the geometry of the roads, but I wonder if there's a way to get other data about roads, like number of lanes (left and right), traffic directions, material/friction coefficient, etc.

masskro0, 2020-08-17:
The scenario class offers the method _get_roads_list(). It returns a list of roads as dict type containing material, rid, drivability, one_way, flip_direction, looped, smoothness, break_angle, texture_length, render_priority, improved_spline, nodes, road_id. 
The BeamNG class offers the method get_roads(), It returns a dict with the rids as keys and drivability, oneWay, flipDirection, lanesRight, lanesLeft as values.

pascale-commits, 2020-09-04:
I took a look at the code and found that it is as follows:

For roads defined in the game, you can retrieve the following properties of a DecalRoad through the get_roads() function:
- 'drivability'
- 'lanesLeft'
- 'lanesRight'
- 'oneWay'
- 'flipDirection'

The other properties masskro0 mentioned can only be accessed through _get_roads_list() if you choose to create an additional road for the scenario with the help of the add_road() method and Road class. 

We do not recommend using private functions, such as _get_roads_list()

Hope that helps and thanks to @masskro0 for helping out.

closed 2020-10-19 by pascale-commits