BeamNG/BeamNGpy #69: Getting current road

ASionville, 2020-08-10:
Hey !

Is there a way to get the Road object corresponding to the road a car is on ?? (it's in order to get the bouding box of this road and then make the car follow the path "manually" with a script that control the steering)

pascale-commits, 2020-08-12:
Hi! 

Did you already take a look at the road_definition.py provided in the examples?

pascale-commits, 2020-08-12:
Ok, so as it turns out there is also a get_roads function in the BeamNGpy class you could use. The function returns a list of tuples that make up the road network. Each tuple consists of three points that make up the left-most point middle point and right-most point of the road.

I'm going to write a jupyter notebook to better visualize how the road network can be used and to give an example on how to use the return values of get_roads.

pascale-commits, 2020-08-19:
You can find the [example ](https://github.com/BeamNG/BeamNGpy/blob/dev/examples/access_road_network.ipynb)in the dev branch. Hope that helps.

closed 2020-08-19 by pascale-commits