BeamNG/BeamNGpy #70: get_road_edges returns an error

masskro0, 2020-08-17:
When I call the get_road_edges method of the Beamng class, I'm getting the following error in the BeamNG console:

030.599|E|GameEngineLua:Exception|[string "lua/ge/extensions//util/researchGE.lua"]:446: attempt to call method 'getEdgesTable' (a nil value)
=============== Stack Traceback >> START >>
(1) method C function 'getEdgesTable'
(2) Lua upvalue 'checkMessage' at line 446 of chunk '"lua/ge/extensions//util/researchGE.lua"]'
  Local variables:
  msg = table: 0x02261a676bf8  {type:GetDecalRoadEdges, road:main_road}
  roadID = string: "main_road"
  response = table: 0x02266f8cd5f8  {type:DecalRoadEdges}
  road = userdata: 0x02266f8cd6a8
  edges = table: 0x02266f8cd6c0  {}
  (*temporary) = C function: ipairs
  (*temporary) = userdata: 0x02266f8cd6a8
  (*temporary) = nil
  (*temporary) = table: 0x02261a676bf8  {type:GetDecalRoadEdges, road:main_road}
  (*temporary) = userdata: 0x02266f8cd6a8
  (*temporary) = string: "éñtype▒GetDecalRoadEdgesñroad®main_road"
  (*temporary) = table: 0x02261a676ad8  {i:40, underflow:function: 0x02261a676b88, s:éñtype▒GetDecalRoadEdgesñroad®main_road (more...)}
  (*temporary) = table: 0x02261a676bf8  {type:GetDecalRoadEdges, road:main_road}
  (*temporary) = number: 40
  (*temporary) = number: 39
  (*temporary) = string: "attempt to call method 'getEdgesTable' (a nil value)"
(3) Lua function '?' at line 106 of chunk '"lua/ge/extensions//util/researchGE.lua"]'
  Local variables:
  dt = number: 0.151158
(4) Lua field 'hook' at line 443 of chunk '"lua/common/extensions.lua"]'
  Local variables:
  func = string: "onPreRender"
  funcList = table: 0x02261a6862e0  {1:function: 0x02265feb4240, 2:function: 0x02265fe8b120, 3:function: 0x02265ffd21d8 (more...)}
  (for index) = number: 10
  (for limit) = number: 10
  (for step) = number: 1
  i = number: 10
(5) Lua function 'luaPreRender' at file 'lua/ge/main.lua:281' (best guess)
  Local variables:
  dtReal = number: 0.151158
  dtSim = number: 0.05
  dtRaw = number: 0.0331754
--------------- << END <<

I'm calling the method get_roads and give the first the road rid as an argument. I'm using BeamNGpy 1.15 and BeamNG revision #45.

pascale-commits, 2020-09-04:
The dev branch has an example of using this function: [access_road_network.ipynb](https://github.com/BeamNG/BeamNGpy/blob/dev/examples/access_road_network.ipynb). It should work with your beamngpy version.
Can you try it and see whether it works?

masskro0, 2020-09-08:
Sadly, I'm getting the same error message. I assume this has something to do with my configuration, I will try this in the future with the newest versions of BeamNGpy and BeamNG.research.

masskro0, 2020-10-27:
I tried this with the newest BeamNGpy (1.18) and BeamNG.research (1.7.0.0) version and this method works fine. However, I couldn't fix this with the previous versions. I'm closing this issue.

closed 2020-10-27 by masskro0