BeamNG/BeamNGpy #71: get_vehicle_bbox method freezes

masskro0, 2020-08-17:
When I call the get_vehicle_bbox method of the Beamng class, then I'm not getting any return. The data is sent but nothing can be received. I'm calling this function after polling the sensors and updating the vehicle state. There is neither an error in Python nor in the BeamNG console. Does anyone know what's the issue here?

ASionville, 2020-08-19:
Can you show us your script?

masskro0, 2020-08-19:
```
from beamngpy import BeamNGpy, Scenario, Vehicle

bng = BeamNGpy('localhost', 64256)
scenario = Scenario('GridMap', 'example')
ego = Vehicle('ego', model='etk800')
scenario.add_vehicle(ego, pos=(0, 0, 0), rot=(0, 0, 45))
scenario.make(bng)
bng.open()
bng.load_scenario(scenario)
bng.start_scenario()

while True:
    ego.update_vehicle()
    bboxes = bng.get_vehicle_bbox(ego)
    print(bboxes)
```

ASionville, 2020-08-19:
Does the game still run or does it freeze?

If it freezes, try adding this before "load_scenario :
`bng.set_steps_per_second(60)
`

masskro0, 2020-08-19:
No, it doesn't freeze. But every code after bng.get_vehicle_bbox() won't be executed since the socket is waiting for a response, which will never come.

ASionville, 2020-08-19:
Another idea, try putting a small delay before starting the while loop, so the game has a little bit of time to set everything up before you bother it 😄

alessiogambi, 2020-08-19:
Does the scenario define any roads? It does not seems so... Not that this
should matter for getting vehicles bbox.

pascale-commits, 2020-09-08:
I have tried your code on my machine and it works just fine. 
Does it also freeze if the code in the while loop is executed only once? 
Perhaps running a while true loop is too much for your current hardware setup.

Another approach could be using `bng.pause()` before the loop and then `bng.step(amount_of_steps)` in your loop to allow the simulation to continue.
An example using this functionality can be found in [west_coast_random](https://github.com/BeamNG/BeamNGpy/blob/master/examples/west_coast_random.py).

masskro0, 2020-09-08:
I've tried all of your suggestions but sadly nothing works. The simulation is still freezing.

pascale-commits, 2020-09-08:
Please try it with the recommended setup and let me know if the problems persists :)

masskro0, 2020-10-27:
I tried this with the newest BeamNGpy (1.18) and BeamNG.research (1.7.0.0) version and this method works fine. However, I couldn't fix this with the previous versions. I'm closing this issue.

closed 2020-10-27 by masskro0