BeamNG/BeamNGpy #72: Can not interact with the game

ASionville, 2020-08-25:
Hi there!

I don't know if the problem comes from BeamNG-Py or BeamNG Research, but when I start a scenario (whatever it is), I can not interact at all with the game (mouse / keyboard / controller doing nothing)

Does anyone have this issue too?

masskro0, 2020-08-26:
Do you have an AI running? If so, the AI must pass all waypoints until you can control it. You can also add another vehicle without waypoints and control that or remove the waypoints from the AI.

ASionville, 2020-08-26:
Here is the script I use, you'll find all your answers in it :

```python
import sys
from time import sleep
import numpy as np
from scipy import interpolate
from beamngpy import BeamNGpy, Scenario, Road, Vehicle, setup_logging

keyboard = Controller()

def main():
    setup_logging()

    beamng = BeamNGpy('localhost', 64256, home="../../research/trunk")
    scenario = Scenario('west_coast_usa', 'ai_sine')

    vehicle = Vehicle('ego_vehicle', model='etk800', licence='AI')

    orig = (-769.1, 400.8, 142.8)

    scenario.add_vehicle(vehicle, pos=orig, rot=(0, 0, 180))
    scenario.make(beamng)

    bng = beamng.open(launch=True)
    try:
        bng.load_scenario(scenario)

        bng.start_scenario()

        while True:
            bng.step(60)
    finally:
        bng.close()

if __name__ == '__main__':
    main()```

masskro0, 2020-08-28:
I just tried your code and I could control the car manually and look around as well. I tested it in BeamNGpy 1.15 and BeamNG version 1.9 revision 45.

ASionville, 2020-08-28:
I work with BeamngPy 1.16.5 and Beamng.research 1.7.0.0
Where can I find the 1.9?

Also, I found that when I plug/unplug a controller, then I can control the game (if not, even the escape key doesn't do anything)

masskro0, 2020-08-28:
I'm using an older version than yours. Version 1.9 gets printed on the Python console but I assume it's research version 1.5 (if I'm not mistaken, pretty confusing). You are using revision 47 while I'm using revision 45.
Just some ideas: check if the throttle is assigned to any key (default should be the up arrow), check if any command blocks further computation by debugging, try starting BeamNG manually and load a scenario (without Python).
Also, is this your full code? There are many imports which aren't used and since I don't know how the rest of your code looks like I can only make assumptions of the code that you've provided.

ASionville, 2020-08-28:
1 - I can't check the keys as I can't do anything, even move the camera or trigger the UI
2 - This is my full code, I don't care of the imports I don't use (was only supposed to be a quick test)

pascale-commits, 2020-10-08:
I also tried your code and it worked fine on my machine. 
Can you paste the log, or preferably send it to me by mail?
It's pmaul(at)beamng.gmbh

On the versioning:
You are talking about three different versioning types:
1. the internal versioning system depending on the build, which is printed on the console and log
2. the svn versioning, which is the revision number
3. the versioning of the BeamNG.research build, where the latest is 1.7.0.0 which uses revision 47
I think its safe to ignore number 1.

ASionville, 2020-10-11:
Email sent

pascale-commits, 2021-01-07:
Hi, excuse the delay.

This might be a bug that can be worked around at the moment by disabling all online features. To do so you can simply delete the BeamNG.research directory in your ` C:\Users\<yourPCUserName>\Documents` folder. When starting up BeamNG through a script you do not need to do anything, but if the first startup is manual, then choose 'No, thanks' in the 'Enable Online Features?' dialogue.

closed 2025-01-27 by aivora-beamng