BeamNG/BeamNGpy #73: Light flares/sprites visible in Camera sensor despite no car is there

alessiogambi, 2020-09-08:
We noticed that the camera sensor while the ego-car is breaking (or going backward) produces images where a red/white light is visible (biasing AI/DL predictions).

You can see the attached images for an example. Note I clipped the images to fit in this page.

[image]
[image]

These images have been collected using the following configuration:
>> I'm using Python 3.7, Windows 10 64 Bit, BeamNGpy 1.15, BeamNG.research unlimited revision #45 (#47 is the newest)

alessiogambi, 2020-09-08:
@masskro0 might be interested (as he created those images in the first place ;) )

pascale-commits, 2020-09-08:
Could you post the code that produced these images?

alessiogambi, 2020-09-08:
Guess that @masskro0 can do that 👍

masskro0, 2020-10-22:
Sorry for the late reply. Here is the code that I used:
```python
# Setup camera.
direction = (0, 1, 0)
fov = 90
resolution = (1280, 720)
x, y, z = -0.3, 2.1, 1
camera = Camera((x, y, z), direction, fov, resolution, colour=True, depth=True, annotation=True)
ego.attach_sensor("camera", camera)
beamng = BeamNGpy('localhost', 64286)
bng = beamng.open()
bng.load_scenario(converter.scenario)
bng.start_scenario()
while True:
    sensors = bng.poll_sensors(ego)
    ego.update_vehicle()
    img = sensors["camera"]["colour"].convert("RGB")
    filename = label + '_{}.png'.format(time())
    file_path = join(image_dir, filename)
    img.save(file_path)
```