BeamNG/BeamNGpy #74: Load an existing scenario and attach to existing cars

alessiogambi, 2020-09-11:
In the code base, there are few examples on creating new scenarios and adding cars to them. But what if one has to work on an existing scenario and plot data from it? So a scenario where roads and vehicles have been already defined in the `.prefab`

alessiogambi, 2020-09-11:
Actually, I might have already did it (so I share my code below). The code assumes a running simulation of a car with id `ego-car`

```
 self.vehicle = Vehicle('ego-car')
electrics = Electrics()
self.vehicle.attach_sensor('electrics', electrics)
# Connect to running beamng
self.bng = BeamNGpy('localhost', 64256)
self.bng = self.bng.open(launch=False)
# Put simulator in pause awaiting while planning the driving
self.bng.pause()
# Connect to the existing vehicle (identified by the ID set in the vehicle instance)
self.bng.set_deterministic()  # Set simulator to be deterministic
self.bng.connect_vehicle(self.vehicle)
assert self.vehicle.skt
# Get Initial state of the car. This assumes that the script is invoked after the scenario is started
self.bng.poll_sensors(self.vehicle)
```

pascale-commits, 2020-10-08:
Hi! 

Just to make sure I've understood the question: do you want to load the game and a scenario through an existing prefab and then connect with python through the BeamNG lib to the existing scenario?

alessiogambi, 2020-10-08:
Yes. 
My use case is the following: A program creates and launches a scenario and `researchGE` is loaded; then, at some point, the second program (typically, an AI) attaches to the running scenario and does something (e.g., drive a car).

An extension might be that more than one program (in turn) attaches to the running scenario.

pascale-commits, 2020-10-09:
It is possible to load existing scenarios by using the scenario.find(...) instead of scenario.load(...). But connecting to individual vehicles is not possible as of now but we might add that in the future, see issue #49 for that.

closed 2020-10-09 by pascale-commits