BeamNG/BeamNGpy #76: BeamNG Keyboard input not working?

umerhamdan10, 2020-09-20:
My keyboard inputs are not working like view_camera, drive car manually etc

ASionville, 2020-09-22:
I got the same, do you have any controller plugged in ?

pascale-commits, 2020-10-08:
Can you post the code you're using and post or send the log file to my mail address?
It's pmaul(a)beamng.gmbh.

pascale-commits, 2020-10-08:
And please specify what version you are using of BeamNG.research and BeamNGpy

umerhamdan10, 2020-10-08:
sorry i did not solve this problem

pascale-commits, 2020-10-09:
Yes, I'm here to help solving this problem, I'm from the company :)

pascale-commits, 2021-01-07:
I've addressed one possible solution [here](https://github.com/BeamNG/BeamNGpy/issues/72#issuecomment-756188305).
Please reopen this issue if your experience differs from the one addressed in issue 72.

closed 2021-01-07 by pascale-commits