BeamNG/BeamNGpy #78: Connecting to BeamNG Game (not Research edition)

Seiwert, 2020-10-14:
Hello,

I'm trying to connect to the BeamNG game instead of the Research edition by following the instructions laid out in the README. However, I keep getting a socket timeout error when calling _bng.open()_. Here is a screenshot of my terminal showing what I've done and the error:
[image]

I've also tried a few other things such as double checking the firewall and trying to connect with the game open and passing the "_launch=False_" argument to _bng.open_

Any guidance here would be greatly appreciated or if there is any other further documentation I should be referring to. Thanks!

Palculator, 2020-10-14:
Yo, couple of questions:

- Does the game launch at all; as in, does the game launch and the Python process is simply sitting there waiting for it to respond but times out instead?
- Is a mod containing required Lua files being created under `Documents/BeamNG.drive/mods/BeamNGpy.zip`?
- Similarly, is the scenario you're making placed in `Documents/BeamNG.drive/levels/west_coast_usa/scenarios/example.*`?

If no BeamNG.drive process is spawned at all, you could maybe add a simple print or check in a debugger what kind of call the library is issuing in this function: https://github.com/BeamNG/BeamNGpy/blob/master/src/beamngpy/beamng.py#L250

You can also import the `setup_logging()` function from here and call it with a desired log file name before creating a `BeamNGpy` instance to get more verbose output in general: https://github.com/BeamNG/BeamNGpy/blob/master/src/beamngpy/beamng.py#L56

Seiwert, 2020-10-14:
To answer your questions:

1. The game does not launch at all. It does as you describe and sits there waiting for a response but times out.
2. I see the file you describe being created in my ```%USERPROFILE%/Documents/``` directory. Not sure if the multi drive paths are an issue but worth mentioning this goes to my C: drive while the game is on my D: drive
3. I see the folder you mention being created but the contents are rather slim: 
[image]

I'll run some more tests with the additional logging statements you called out.

Seiwert, 2020-10-14:
So here is some more information I collected:

- Here is a run with just using setup_logging() called from the pip install:
[image]
Not too much additional information beyond making the subprocess call the prime suspect.

- So moving onto a test and printing out the _call_ variable:
[image]
Nothing immediately stands out here to me except it seems to confirm that something is going wrong with the subprocess call. Thoughts @Palculator ?

Seiwert, 2020-10-14:
As an additional sanity check, I ran the following test:

```python
import subprocess
test_process = subprocess.Popen(['D:\\Program Files (x86)\\Steam\\steamapps\\common\\BeamNG.drive\\Bin64\\BeamNG.drive.x64.exe'])
```
This launched the game without any issues, so wondering if it's one of the launch arguments being passed in.

Palculator, 2020-10-14:
Yeah, it does look fine to me. What complicates things further is that I also grabbed the Steam release and tested it myself; it worked fine, using the same parameters and even a similar setup where I have the game on the `D:\` drive and my userpath on `C:\`:

```
(.venv) PS C:\Users\Marc\Documents\Code\BeamNG\BeamNGpy> python .\examples\ai_line.py
2020-10-14 10:30:29,610 INFO     Started BeamNGpy logging.
2020-10-14 10:30:29,610 DEBUG    Determined BeamNG.* binary to be: D:\SteamLibrary\steamapps\common\BeamNG.drive\Bin64\BeamNG.drive.x64.exe
2020-10-14 10:30:29,660 INFO     Opening BeamNPy instance...
2020-10-14 10:30:29,665 INFO     Started BeamNGpy server on localhost:64256
2020-10-14 10:30:29,666 DEBUG    Starting BeamNG process: ['D:\\SteamLibrary\\steamapps\\common\\BeamNG.drive\\Bin64\\BeamNG.drive.x64.exe', '-console', '-rport', '64256', '-rhost', 'localhost', '-nosteam', '-physicsfps', '4000', '-lua', "registerCoreModule('util/researchGE')", '-userpath', 'C:\\Users\\Marc\\Documents\\BeamNG.drive']
2020-10-14 10:30:39,357 DEBUG    Connection established. Awaiting "hello"...
2020-10-14 10:30:39,359 INFO     Started BeamNGpy communicating on ('127.0.0.1', 51380)
2020-10-14 10:30:57,890 DEBUG    Starting vehicle server for V:ego_vehicle on: localhost:64257
```

You can try dropping some of the parameters like `-nosteam` and `-physicsfps 4000` (this one doesn't even do anything in the Steam version afaik) but they seem to be okay for me, at least.

I hope it's not something as silly as the spaces or the (x86) in the path 🙃

Seiwert, 2020-10-14:
So some more testing reveals that I got it to work with this call:
```python
subprocess.Popen(['D:\\Program Files (x86)\\Steam\\steamapps\\common\\BeamNG.drive\\Bin64\\BeamNG.drive.x64.exe', '-console', '-rport', '64256', '-rhost', 'localhost', '-nosteam', '-physicsfps', '4000', '-lua', "registerCoreModule('util/researchGE')", '-userpath'])
```
The raw output from the library that did not work was this:
```
['D:\\Program Files (x86)\\Steam\\steamapps\\common\\BeamNG.drive\\Bin64\\BeamNG.drive.x64.exe', '-rport', '64256', '-rhost', 'localhost', '-physicsfps', '4000', '-lua', '-userpath', 'C:\\Users\\Adam Seiwert\\Documents\\BeamNG.drive']
```
So the error seems to lie with the _userpath_ . Maybe it is the space in my username?

Also, no comment on the poorly named x86 program files folder.... :sweat_smile:

Palculator, 2020-10-14:
Interesting. You can pass the `BeamNGpy` constructor a `user='D:/path/to/custom/userpath'` parameter to override the default one. This will make it start off fresh, so any mods you've previously installed in your regular userpath would not be there and load times will be a bit slow the first time as caches are generated, but if that works, we got our culprit. Note: The mod containing required Lua files is created & copied to the userpath by the library, so no need to worry about that.

Seiwert, 2020-10-14:
The custom user path did the trick! It is definitely an issue with the space in the userpath. I did a test with a custom userpath of _BEAMNGTEST_ and _BEAMNGTEST WITHSPACE_ and the "withspace" one recreated the issue.

Thanks so much for the help, @Palculator

closed 2020-10-14 by Seiwert

Palculator, 2020-10-14:
Neat, I guess I'll at least put that in the docs or ideally find a way to fix it such that it works with spaces. Thanks for the report!