BeamNG/BeamNGpy #79: Linux Support

Seiwert, 2020-10-19:
### TL;DR: Any possible conversion for mmap sensor declarations to work on Unix?

Feel free to immediately close or delete this ticket as it's more of an inquiry and less of an issue.

To start things off, I've been looking into the possibility of running BeamNG/BeamNG Research in Linux/Ubuntu vs Windows (a lot more tools available for additional development in Linux vs. Windows). The setup for all this research is a Ubuntu 18.04 machine. My initial investigation led me to looking into running BeamNG with Steam utilizing Proton. This work fabulously and the game ran immediately out of the box. Going off of this, I attempted to run BeamNG Research with Proton as well utilizing Steam's ability to run non Steam games. This did not work so well and I abandoned this and just went straight to using Wine. Wine was able to run the research edition fine with only a few missing UI elements but the performance was good a no other issues noted. 

Knowing this, I went on to making some small modifications to BeamNGpy to support Wine usage. I got that straightened out but I've ran into a problem that is leaving me rather stumped. Looking through the code, I see that shared allocated memory with _mmap_ is used to share sensor data between the game engine and Python (e.g. https://github.com/BeamNG/BeamNGpy/blob/master/src/beamngpy/sensors.py#L439)

Now, this is totally fine and makes perfect sense, however, there is a key issue being the [difference in how Window's handles mmap](https://docs.python.org/3.7/library/mmap.html). Given this and not really having any insight into how the memory scheme is being handled from engine to Python (minus what appears to be the use of tagname usage - not available in unix) , I did a pretty big stupid and just assigned an anonymous map. _**Shockingly**_ I got an access error when the sensor polling cycle came up. However, as you can see in the attached screenshot, it seems all other elements are in working order (note: not all textures had loaded in before first polling cycle so ignore that):
[image]

So I guess my underlying question is there any possible conversion path to get these mmap declarations working with Unix compatibility or is that too big of an ask on the game engine side?

Thanks!

Palculator, 2020-10-20:
Hey, Linux support for academia & research is something we are planning since, as you point out, it's a more natural fit there, but no promises or an ETA yet. Regarding memory mapping, I reckon it boils down to the C/C++ side of things relying on Windows-specific functions to do that, namely `OpenFileMappingA` and `MapViewOfFile` which I guess are not properly supported by Wine/Proton (?). Depending on how important performance in your particular case is, we could re-introduce the old way of transferring large data between Python and BeamNG: the sockets. I switched to shared memory because it's **a lot** faster but good old robust sockets could be supported as a fallback option.

Seiwert, 2020-10-22:
I don't know if I necessarily follow what you're referring to with "OpenFileMappingA" and "MapViewOfFile". I'm assuming this is pertaining more to how you're doing this on the game engine side of things (CryEngine?). 

I guess what I was trying to point to with the memory mapping issue was pertaining more to the exact API difference between Windows and Unix. From the docs:

> (Windows version) Maps length bytes from the file specified **by the file handle fileno,** and creates a mmap object
> (Unix version) Maps length bytes from the file specified **by the file descriptor fileno**, and returns a mmap object. 

So referring [back to the code I linked](https://github.com/BeamNG/BeamNGpy/blob/master/src/beamngpy/sensors.py#L439), it seems as though within BeamNGpy the fileno is set to 0 which indicates a max length map correlating to the size of the file (Windows based). So this leads me to believe the mapping is created using the tagname parameter. This in turn leads to the issue I'm seeing and which the Python docs calls out:
> Avoiding the use of the tag parameter will assist in keeping your code portable between Unix and Windows.

So I guess my initial statement was more or less banking on the hope of some easy interface from the engine back to this memory map without relying upon the tag parameter.

Referencing to your statement about socket usage though, I think that could also be a good solution. From what I've seen so far, this is actually more common than any type of memory sharing (e.g. Gazebo, AirSim, Isaac, etc..). Is the socket interface something that is available now or in a previous release?

Palculator, 2020-10-23:
Yo, the game engine side of things relies on functions that in turn rely on those tag names and not `fileno`, that's what's probably breaking things since I'm not sure how Wine handles that. I've committed a simple change that creates tempfiles backing the shmems to get a valid `fileno` in a `mmap_tempfiles` branch you can give a go, maybe that's enough to make it work under Wine/Proton: https://github.com/BeamNG/BeamNGpy/tree/mmap_tempfiles

Similarly, for socket communication there's a test branch here: https://github.com/BeamNG/BeamNGpy/tree/puresockets
It's kind of hacky right now as I simply rewrote existing code to use some of the older functions that haven't been in use for ages. The basic unit tests work, but to make it cleaner and a proper fallback feature I'd like to do some restructuring in all components (C++, Lua, and Python.) Let's see if it's worth it by checking if it even works first. Fair warning: the reason I switched to shared memory back then was sockets being too slow to transfer in bulk like that, especially for the Lidar.

PS: The engine hasn't been CryEngine since the very early stages of the project. It was later switched to be based on Torque3D but has now been ship-of-theseus'd into its own thing, pretty much. 😉

Seiwert, 2021-02-09:
Hey! I'm so so sorry on the delayed response here. Definitely still interested and will try to be more responsive.

So to address the tempfiles test, I ran that branch and saw the following error:
```bash
2021-02-08 21:26:59,787 ERROR    Uncaught exception: 
Traceback (most recent call last):
  File "../examples/lidar_tour.py", line 95, in <module>
    main()
  File "../examples/lidar_tour.py", line 57, in main
    vehicle.attach_sensor('lidar', lidar)
  File "/home/adam/Documents/git/BeamNGpy/src/beamngpy/vehicle.py", line 161, in attach_sensor
    sensor.attach(self, name)
  File "/home/adam/Documents/git/BeamNGpy/src/beamngpy/sensors.py", line 456, in attach
    Lidar.shmem_size, self.handle)
TypeError: an integer is required (got type str)
2021-02-08 21:26:59,795 WARNING  sys:1: ResourceWarning: unclosed file <_io.BufferedRandom name=4>
```
This makes sense given the reference to [self.handle in the mmap call](https://github.com/BeamNG/BeamNGpy/blob/mmap_tempfiles/src/beamngpy/sensors.py#L456). Adjusting to the Linux happy version of mmap call with this -> 
```python
self.shmem = mmap.mmap(self.file.fileno(),0)
```

then led to this error ->

```bash
2021-02-08 21:36:44,121 ERROR    Uncaught exception: 
Traceback (most recent call last):
  File "../examples/lidar_tour.py", line 95, in <module>
    main()
  File "../examples/lidar_tour.py", line 57, in main
    vehicle.attach_sensor('lidar', lidar)
  File "/home/adam/Documents/git/BeamNGpy/src/beamngpy/vehicle.py", line 161, in attach_sensor
    sensor.attach(self, name)
  File "/home/adam/Documents/git/BeamNGpy/src/beamngpy/sensors.py", line 455, in attach
    self.shmem = mmap.mmap(self.file.fileno(),0)
ValueError: cannot mmap an empty file
2021-02-08 21:36:44,130 WARNING  sys:1: ResourceWarning: unclosed file <_io.BufferedRandom name=4>
```
So unless you have some additional ideas here this may be a dead end.

===========================================================
Moving onto the PureSocket branch that appeared as though it was going to work. However, apparently since I've last worked on this I've managed to somehow nuke my Wine installation or Nvidia driver or who knows what else and I keep seeing this error over and over again with just a black screen
```bash
00c4:fixme:d3d11:d3d11_immediate_context_VSGetShader Dynamic linking not implemented yet.
00c4:fixme:d3d11:d3d11_immediate_context_PSSetShader Dynamic linking is not implemented yet.
```
This is probably something just on my system and I'm still debugging this. 

But while we're talking about bringing back socket communication and shared memory, if I could make a future development suggestion, I would recommend you align to a more common middleware so that you don't have to worry about any of it. One really nice pairing potential in my opinion would be to follow suite with the Robot Operating System and utilize DDS https://design.ros2.org/articles/ros_on_dds.html
I won't go into too much more detail right now but if you're curious I could talk with you more about it.

Thanks!

closed 2025-01-27 by aivora-beamng