BeamNG/BeamNGpy #83: Load different configuration of vehicle

masskro0, 2020-10-28:
I'm struggling to load a configuration of a vehicle via Python, e.g. using the etk856td_M instead of the default one. Is there any way to do this in Python? The method get_part_config() in the vehicle class gives me the output
```python
{'vars': {}, 'parts': {}, 'pcPath': ''}
```
So I thought I could just use the pcPath argument to pass the path to the .pc file of the configuration like this:
```python
cfg = {'vars': {}, 'parts': {}, 'pcPath': "D:\\Program Files (x86)\\BeamNG\\vehicles\\etk800\\etk856td_M.pc"}
ego.set_part_config(cfg)
```
However, I'm getting a huge error message in the console:
```
017.651|E|libbeamng.lua.V|*** FATAL LUA ERROR: lua/common/utils.lua:927: attempt to call a nil value
017.652|E|libbeamng.lua.V|=============== Stack Traceback >> START >>
017.653|E|libbeamng.lua.V|(1) Lua global 'deserialize' at file 'lua/common/utils.lua:927'
017.653|E|libbeamng.lua.V|  Local variables:
017.654|E|libbeamng.lua.V|  s = string: "_kv{\"bdebug\";_kv{\"vehicle\";_kv{\"nodeText_modes\";{\"off\",\"names\",\"numbers\",\"names+numbers\",\"weights\",\"materials\"},\"beamVis\";\"off\",\"meshVisibility\";100,\"beamVisAlpha\";1,\"collisionTriangle\";false,\"cog\";1,\"cog_modes\";{\"off\",\"on\",\"nowheels\"},\"aero_modes\";{\"off\",\"drag+lift\",\"aoa\",\"combined\"},\"nodeVis\";\"off\",\"objectData\";false,\"nodeText\";\"off\",\"aero\";\"off\",\"beamVis_modes\";{\"off\",\"simple\",\"type\",\"with broken\",\"broken only\",\"stress\",\"deformation\",\"breakgroups\",\"deformgroups\"},\"nodeVis_modes\";{\"off\",\"simple\",\"weights\",\"velocities\",\"forces\",\"density\"},\"tireContactPoint\";false,\"aerodynamicsScale\";0.1},\"fov\";80,\"vehicleDebugVisible\";false,\"terrain\";_kv{\"staticCollision\";false,\"groundmodel\";false},\"renderer\";_kv{\"disableShadows\";false,\"showFps\";false,\"wireframe\";false,\"visualization\";\"None\",\"boundingboxes\";false},\"physicsEnabled\";true},\"energyStorage\";_kv{\"mainTank\";_kv{\"remainingVolume\";49.9998}},\"controller\";_kv{\"twoStepLaunch\";_kv{\"state\";\"deactivated\",\"rpm\";1700},\"esc\";_kv{\"escConfigKey\";1}},\"controller/vehicleController\";_kv{\"previousGearboxBehavior\";\"arcade\"},\"sounds\";_kv{\"createSFXSource\";nil,\"refNode\";387,\"FMODHzToHz\";nil,\"uiDebugging\";false,\"playSoundOnceFollowNode\";nil,\"disableOldEngineSounds\";nil,\"reset\";nil,\"setUIDebug\";nil,\"playSoundOnceAtNode\";nil,\"bodyCollision\";nil,\"playSound\";nil,\"getBeamSounds\";nil,\"engineNode\";244,\"hzToFMODHz\";nil,\"updateGFX\";nil,\"init\";nil,\"onDeserialized\";nil,\"usesOldCustomSounds\";false,\"createSoundscapeSound\";nil},\"partmgmt\";_kv{\"config\";_kv{\"vars\";{},\"parts\";{},\"pcPath\";\"D:\\Program Files (x86)\\BeamNG\\vehicles\\etk800\\etk856td_M.pc\"}},\"recovery\";_kv{\"safePoints\";_kv{\"home\";_kv{\"dirFront\";float3(0.5,0.866025,0),\"pos\";vec3(4,2,0),\"dirUp\";float3(0,0,1)}},\"startRecovering\";nil,\"stopRecovering\";nil,\"loadHome\";nil,\"recoverInPlace\";nil,\"saveHome\";nil,\"loadPoint\";nil,\"savePoint\";nil,\"updateGFX\";nil,\"onDeserialized\";nil,\"init\";nil,\"clear\";nil,\"recoveryPoints\";_kv{\"_serialize\";true,1;_kv{\"dirFront\";float3(0.5,0.866025,0),\"pos\";vec3(4,2,0),\"dirUp\";float3(0,0,1)},\"tail\";1,\"head\";0,}},\"ai\";_kv{\"stateChanged\";nil,\"driveUsingPath\";nil,\"dumpCurrentRoute\";nil,\"startFollowing\";nil,\"scriptState\";nil,\"setAggressionMode\";nil,\"setAggression\";nil,\"cutOffDrivability\";0,\"scriptStop\";nil,\"setState\";nil,\"onDeserialized\";nil,\"driveInLane\";nil,\"setVehicleDebugMode\";nil,\"setCutOffDrivability\";nil,\"setTargetObjectID\";nil,\"setSpeed\";nil,\"setTarget\";nil,\"setMode\";nil,\"setScriptDebugMode\";nil,\"stopFollowing\";nil,\"reset\";nil,\"setAvoidCars\";nil,\"updateGFX\";nil,\"resetLearning\";nil,\"debugMode\";\"off\",\"mode\";\"disabled\",\"stopRecording\";nil,\"getState\";nil,\"startRecording\";nil,\"debugDraw\";nil,\"spanMap\";nil,\"extAggression\";2,\"setSpeedMode\";nil,\"init\";nil,\"isDriving\";nil},\"extensions\";_kv{\"loadedModules\";_kv{\"researchVE\";\"researchVE\",\"core_quickAccess\";\"core/quickAccess\",\"core_performance\";\"core/performance\"}}}"
017.655|E|libbeamng.lua.V|  (*temporary) = nil
017.655|E|libbeamng.lua.V|  (*temporary) = string: "[string \"return _kv{\"bdebug\";_kv{\"vehicle\";_kv{\"nodeTe...\"]:1: invalid escape sequence near '\"D:'"
017.657|E|libbeamng.lua.V|  (*temporary) = string: "return _kv{\"bdebug\";_kv{\"vehicle\";_kv{\"nodeText_modes\";{\"off\",\"names\",\"numbers\",\"names+numbers\",\"weights\",\"materials\"},\"beamVis\";\"off\",\"meshVisibility\";100,\"beamVisAlpha\";1,\"collisionTriangle\";false,\"cog\";1,\"cog_modes\";{\"off\",\"on\",\"nowheels\"},\"aero_modes\";{\"off\",\"drag+lift\",\"aoa\",\"combined\"},\"nodeVis\";\"off\",\"objectData\";false,\"nodeText\";\"off\",\"aero\";\"off\",\"beamVis_modes\";{\"off\",\"simple\",\"type\",\"with broken\",\"broken only\",\"stres
017.657|E|libbeamng.lua.V|*** while executing this line: importPersistentData('_kv{"bdebug";_kv{"vehicle";_kv{"nodeText_modes";{"off","names","numbers","names+numbers","weights","materials"},"beamVis";"off","meshVisibility";100,"beamVisAlpha";1,"collisionTriangle";false,"cog";1,"cog_modes";{"off","on","nowheels"},"aero_modes";{"off","drag+lift","aoa","combined"},"nodeVis";"off","objectData";false,"nodeText";"off","aero";"off","beamVis_modes";{"off","simple","type","with broken","broken only","stress","deformation","breakgroups","deformgroups"},"nodeVis_modes";{"off","simple","weights","velocities","forces","density"},"tireContactPoint";false,"aerodynamicsScale";0.1},"fov";80,"vehicleDebugVisible";false,"terrain";_kv{"staticCollision";false,"groundmodel";false},"renderer";_kv{"disableShadows";false,"showFps";false,"wireframe";false,"visualization";"None","boundingboxes";false},"physicsEnabled";true},"energyStorage";_kv{"mainTank";_kv{"remainingVolume";49.9998}},"controller";_kv{"twoStepLaunch";_kv{"state";"deactivated","rpm";1700},"esc";_kv{"escConfigKey";1}},"controller/vehicleController";_kv{"previousGearboxBehavior";"arcade"},"sounds";_kv{"createSFXSource";nil,"refNode";387,"FMODHzToHz";nil,"uiDebugging";false,"playSoundOnceFollowNode";nil,"disableOldEngineSounds";nil,"reset";nil,"setUIDebug";nil,"playSoundOnceAtNode";nil,"bodyCollision";nil,"playSound";nil,"getBeamSounds";nil,"engineNode";244,"hzToFMODHz";nil,"updateGFX";nil,"init";nil,"onDeserialized";nil,"usesOldCustomSounds";false,"createSoundscapeSound";nil},"partmgmt";_kv{"config";_kv{"vars";{},"parts";{},"pcPath";"D:\\Program Files (x86)\\BeamNG\\vehicles\\etk800\\etk856td_M.pc"}},"recovery";_kv{"safePoints";_kv{"home";_kv{"dirFront";float3(0.5,0.866025,0),"pos";vec3(4,2,0),"dirUp";float3(0,0,1)}},"startRecovering";nil,"stopRecovering";nil,"loadHome";nil,"recoverInPlace";nil,"saveHome";nil,"loadPoint";nil,"savePoint";nil,"updateGFX";nil,"onDeserialized";nil,"init";nil,"clear";nil,"recoveryPoints";_kv{"_serialize";true,1;_kv{"dirFront";float3(0.5,0.866025,0),"pos";vec3(4,2,0),"dirUp";float3(0,0,1)},"tail";1,"head";0,}},"ai";_kv{"stateChanged";nil,"driveUsingPath";nil,"dumpCurrentRoute";nil,"startFollowing";nil,"scriptState";nil,"setAggressionMode";nil,"setAggression";nil,"cutOffDrivability";0,"scriptStop";nil,"setState";nil,"onDeserialized";nil,"driveInLane";nil,"setVehicleDebugMode";nil,"setCutOffDrivability";nil,"setTargetObjectID";nil,"setSpeed";nil,"setTarget";nil,"setMode";nil,"setScriptDebugMode";nil,"stopFollowing";nil,"reset";nil,"setAvoidCars";nil,"updateGFX";nil,"resetLearning";nil,"debugMode";"off","mode";"disabled","stopRecording";nil,"getState";nil,"startRecording";nil,"debugDraw";nil,"spanMap";nil,"extAggression";2,"setSpeedMode";nil,"init";nil,"isDriving";nil},"extensions";_kv{"loadedModules";_kv{"researchVE";"researchVE","core_quickAccess";"core/quickAccess","core_performance";"core/performance"}}}')
017.658|E|libbeamng.lua.V|*** in chunk: string
```
I noticed that in partmgmt.lua the variable is not called "pcPath", but "settings". Changing this didn't solve the issue. Maybe this happens because the other arguments were not passed, but I'm uncertain what these arguments should be. I hope someone can help me out with this problem. Thanks!

pascale-commits, 2020-10-28:
I'll look further into this error, but if you want to **set** a specific vehicle configuration you can also use

```
config = 'vehicles/etk800/police.pc'
ego = Vehicle('vehicle', model='etk800', partConfig=config)
```

masskro0, 2020-10-28:
Thank you! I was exactly looking for this! I checked the wrong places at first.

closed 2020-10-28 by masskro0