BeamNG/BeamNGpy #84: Socket will D/C during loading, making the lib useless

DaRealCodeWritten, 2020-10-30:
I can't use the BeamNGpy library since for some odd reason, while loading a scenario, the socket will timeout and the BeamNG.research console will get spammed with trying to connect, then failing, over and over again, is there any way to avoid the socket timing out?

pascale-commits, 2020-10-30:
What is you setup (OS, Beamngpy version, BeamNG.research version), are you using the newest one?
Can you post your code?

alessiogambi, 2020-10-30:
@DaRealCodeWritten Check your firewall configurations as by default might deny any "untrusted" connections, including the ones from/to localhost.

DaRealCodeWritten, 2020-10-30:
@alessiogambi i already allowed Python and Beam through the firewall

pascale-commits, 2021-01-07:
Please post your code and logfiles (you may also send the log files to: pmaul(a)beamng.com).

DaRealCodeWritten, 2021-01-21:
@pascale-commits Apologies for not responding, my PC has been down (and still is). I got beam working properly now though

pascale-commits, 2021-01-22:
Good to hear, do you know what the issue was and how did you resolve it?

closed 2021-03-05 by Palculator