BeamNG/BeamNGpy #85: Example west_coast_random Fails

EthanJamesLew, 2020-11-05:
## Expected Behavior
The example `west_coast_random.py` launches BeamNG, loads a west coast USA scenario, and makes an etk800 vehicle move randomly. Also, an animation is shown in a Matplotlib figure of the attached front and back cameras.

## Current Behavior
BeamNG launches and loads the scenario. The vehicle is stationary, and the output

```
027.363|E|GameEngineLua:Exception|[string "lua/ge/extensions//util/researchGE.lua"]:580: attempt to index field 'Annotation' (a nil value)
=============== Stack Traceback >> START >>
(1) metamethod C function '__index'
(2) Lua upvalue 'checkMessage' at line 580 of chunk 'lua/ge/extensions//util/researchGE.lua'
  Local variables:
  msg = table: 0x0209283b24c8  {type:EngineFlags, flags:table: 0x020919d3f5b0}
  flags = table: 0x020919d3f5b0  {annotations:true}
  (*temporary) = nil
  (*temporary) = string: "I"
  (*temporary) = string: "Setting engine flags!"
  (*temporary) = Lua function '(anonymous)' (defined at line 1 of chunk lua/ge/extensions//util/researchGE.lua)
  (*temporary) = string: "attempt to index field 'Annotation' (a nil value)"
(3) Lua function '?' at line 133 of chunk 'lua/ge/extensions//util/researchGE.lua'
  Local variables:
  dt = number: 0.0200757
(4) Lua field 'hook' at line 496 of chunk 'lua/common/extensions.lua'
  Local variables:
  func = string: "onPreRender"
  funcList = table: 0x02094ab40098  {1:function: 0x0208cfb52a78, 2:function: 0x020895ef6f10, 3:function: 0x0208cf9c5fa0 (more...)}
  (for index) = number: 1
  (for limit) = number: 11
  (for step) = number: 1
  i = number: 1
(5) Lua function 'luaPreRender' at file 'lua/ge/main.lua:295' (best guess)
  Local variables:
  dtReal = number: 0.0200757
  dtSim = number: 0.0165
  dtRaw = number: 0.0228871
  updateUIData = boolean: true
--------------- << END <<
```
is visible in the BeamNG console.

## Steps to Reproduce
1. Run the example
```
cd examples
python west_coast_lidar.py
```
(python setup)
```
conda create -n beamng python=3.7
conda activate beamng
pip install beamng
```

## Context (Environment)
I'm attempting to interact with the vehicle, such as set part config (`Vehicle.set_part_config`) and control (`Vehicle.control`) it.

alessiogambi, 2020-11-06:
Hi,

What versions of beamngpy and Beamng.research are you using?

EthanJamesLew, 2020-11-06:
Hi! Under my "Steps to Reproduce" section, I list

> (beamng environment)
> ```
> Steam Release
> 0.20.2.0.10611
> ```
Is this the correct info?

alessiogambi, 2020-11-06:
Sorry I missed that!

But I see that you use the Steam Release, which I am not entirely sure is compatible with BeamNG.py,
so that might be it.

As a matter of fact, I am only using BeamNG.research/BeamNG.tech.

kris-agile-hardware, 2020-11-06:
@alessiogambi I'm a colleague of @EthanJamesLew and we've been working with Marc and Pascal on various things to support our BeamNG.drive-based automotive simulator. We noted that behavior changes with the upgrade from beamngpy 1.16 to 1.18.

alessiogambi, 2020-11-06:
@kris-agile-hardware this sounds cool. I did not know there were other beamng-based simulators around. Is this an industrial or academic/open source project? We are using Beamng as reference platform for automatically generating tests scenarios.

kris-agile-hardware, 2020-11-06:
@alessiogambi it's somewhat of an intersection of both industrial and academic spheres. I think that's all I can say.

EthanJamesLew, 2020-11-10:
Update: I switch to

(beamng environment)
```
BeamNG.Research 1.7.0.0
```

Using the `beamng` conda environment, I ran `python west_coast_random.py`, and I receive this error
```
030.591|I|libbeamng.lua.V.ResearchVE|Trying to connect to: localhost:64257
032.637|I|libbeamng.lua.V.ResearchVE|Connected!
032.643|E|GameEngineLua:Exception|[string "lua/ge/extensions//util/researchGE.lua"]:340: attempt to call field 'openShmem' (a nil value)
=============== Stack Traceback >> START >>
(1) field C function 'openShmem'
(2) Lua upvalue 'checkMessage' at line 340 of chunk 'lua/ge/extensions//util/researchGE.lua'
  Local variables:
  msg = table: 0x02783efc9ee8  {type:OpenShmem, name:42652.ego_vehicle.front_cam.colour, size:1048576}
  name = string: "42652.ego_vehicle.front_cam.colour"
  size = number: 1.04858e+06
  (*temporary) = nil
  (*temporary) = Lua function '?' (defined at line 336 of chunk lua/ge/extensions//util/researchGE.lua)
  (*temporary) = string: "42652.ego_vehicle.front_cam.colour"
  (*temporary) = number: 1.04858e+06
  (*temporary) = string: "attempt to call field 'openShmem' (a nil value)"
(3) Lua function '?' at line 133 of chunk 'lua/ge/extensions//util/researchGE.lua'
  Local variables:
  dt = number: 0.0191614
(4) Lua field 'hook' at line 496 of chunk 'lua/common/extensions.lua'
  Local variables:
  func = string: "onPreRender"
  funcList = table: 0x0278ebab1b08  {1:function: 0x02783ecc8378, 2:function: 0x027805f117e0, 3:function: 0x027805e29848 (more...)}
  (for index) = number: 1
  (for limit) = number: 11
  (for step) = number: 1
  i = number: 1
(5) Lua function 'luaPreRender' at file 'lua/ge/main.lua:298' (best guess)
  Local variables:
  dtReal = number: 0.0191614
  dtSim = number: 0.0165
  dtRaw = number: 2.05513
  updateUIData = boolean: true
--------------- << END <<
```

CC @alessiogambi @kris-agile-hardware

pascale-commits, 2020-11-11:
Hi Ethan, have you checked whether your license file is up to date and in the appropriate place? It is supposed to be in `C:/Users/<name>/Documents/BeamNG.research/research.key`. If you are using the Steam version, then you need to put it into the corresponding `BeamNG.drive` folder.
If your license has been accepted, then the log file should contain the following message: `BeamNG.research license verified.`

EthanJamesLew, 2020-11-11:
I didn't see the string `BeamNG.research license verified.` in the `beamng.log` file in `BeamNG.research`, so perhaps that's a promising lead for solving this. I did have `research.key` in (for my case) `C:\Users\lew_e\Documents\BeamNG.research` and `C:\Users\lew_e\Documents\BeamNG.drive`. The effect of this file doesn't seem to be none, as this warning is displayed 
```
No BeamNG.research license found at:
"C:\Users\lew_e\Documents\BeamNG.research\research.key",
Please check your confirmation email for the license file and place it there.
```
without it. The game launches when the key is placed there.

pascale-commits, 2020-11-12:
Hmm, did you change your BNG_HOME variable, when switching the version?
Was the game already running when you started the script?
It would be great if you could post the whole log file.

EthanJamesLew, 2020-11-12:
Yes, I changed the BNG_HOME version. I can see the version printed in the BeamNG title bar, too (and it changes when I switch BNG_HOME to the steam release). I am not starting manually, as I see that `beamng.open(launch=True)`. Here is my log file

```
-- Log started - v 1.7.0.0.0 - x64 - build 0 - 2020-11-11 -- 09:29:46 -----
-- private build Marc - Tue Sep 15 11:16:02 2020 
-- Log Format: Time since startup | Message level: D(ebug), I(nfo), W(arning), E(rror), A(lways) | Message
  0.07664|I|StandardMainLoop::init|Virtual Filesystem: user path: C:/Users/lew_e/Documents/BeamNG.research/
  0.07669|D|StandardMainLoop::init| - user path reason: command argument -userpath
  0.07671|D|StandardMainLoop::init| - user mods path: C:/Users/lew_e/Documents/BeamNG.research/mods/
  0.07673|D|StandardMainLoop::init| - game mods path: C:/Users/lew_e/Desktop/BeamNG.research.v1.7.0.0/content/
  0.07686|I|Processor::init|Intel(R) Core(TM) i7-7820HK CPU @ 2.90GHz ~2.90 GHz [HT,] 1 physical, 4 cores, 8 logical
  0.08358|D|main|H2 24F35B91E7C4A258EAD13A8154277B4A5C28
  0.08485|D|ModuleManager::initializeSystem|initializing 69 modules (only displaying anything above 0.100000 s)...
  0.08501|I|SFXFMODProvider::init|SFXFMODProvider - FMOD API version: 2.0.5
  0.16177|A|GELua.lua.utils|"desiredwindowPlacement: 0 1 -1 -1 -1 -1 -70 150 1206 1866"
  0.17226|D|GELua.core_gamestate.gamestate|see if ui is loaded
  0.22083|D|GFXInit::init|GFX Init:
  0.22559|D|_GFXInitReportAdapters| Direct 3D (version 11) device found
  0.22561|D|_GFXInitReportAdapters|   Null device found
  0.22565|D|ModuleManager::initializeSystem|... all modules loaded in 0.140810 seconds
  0.22566|D|Platform::init|Initializing platform...
  0.23761|D|DirectInputManager::enumerateDevices| *** Input devices start ***
  0.24441|D|DirectInputDevice::create|Mouse [instance: Mouse] detected, created as mouse0 (asynchronous)
  0.24445|D|DirectInputDevice::create| - devtype 0x112
  0.24447|D|DirectInputDevice::create| - A mouse or mouse-like device (such as a trackball)
  0.24449|D|DirectInputDevice::create| + 3 axes
  0.24452|D|DirectInputDevice::create| + 5 buttons
  0.24462|D|DirectInputDevice::create|created device 'mouse0', took 0.000s: create:0.005ms, caps:0.002ms, txts:0.157ms, enum:0.034ms, fmt:0.011ms, buff:0.009ms, axis:0.000ms
  0.24586|D|DirectInputDevice::create|Keyboard [instance: Keyboard] detected, created as keyboard0 (asynchronous)
  0.24589|D|DirectInputDevice::create| - devtype 0x413
  0.24592|D|DirectInputDevice::create| - Keyboard or keyboard-like:
  0.24594|D|DirectInputDevice::create|  - IBM PC Enhanced 101/102-key or Microsoft Natural keyboard.
  0.24596|D|DirectInputDevice::create| + 128 buttons
  0.24678|D|DirectInputDevice::create|created device 'keyboard0', took 0.001s: create:0.227ms, caps:0.001ms, txts:0.147ms, enum:0.748ms, fmt:0.028ms, buff:0.000ms, axis:0.000ms
  0.30149|D|DirectInputManager::enumerateDevices| *** Input devices end ***
  0.30156|D|DirectInputDevice::acquire|Not attempting to acquire device mouse0, since game window is not available
  0.30159|D|DirectInputDevice::acquire|Not attempting to acquire device keyboard0, since game window is not available
  0.30169|D|Platform::init|Intel(R) Core(TM) i7-7820HK CPU @ 2.90GHz GenuineIntel x64
  0.30178|D|Platform::init|Microsoft Windows 10 (v10.0) (build 19041), 64-bit
  0.30181|D|Platform::init|Physical memory: 15.96 GB
  0.30186|D|Platform::init|Battery: state: 'high' - online
  0.30188|D|Platform::init|Process Priority: Normal (0x20)
  0.30234|D|CommandListener|CommandListener started
  0.30322|D|TS|command line args:
  0.30326|D|TS| - 0 = C:\Users\lew_e\Desktop\BeamNG.research.v1.7.0.0\Bin64\BeamNG.research.x64.exe
  0.30329|D|TS| - 1 = -console
  0.30332|D|TS| - 2 = -rport
  0.30335|D|TS| - 3 = 64256
  0.30338|D|TS| - 4 = -rhost
  0.30340|D|TS| - 5 = localhost
  0.30344|D|TS| - 6 = -nosteam
  0.30347|D|TS| - 7 = -physicsfps
  0.30349|D|TS| - 8 = 4000
  0.30351|D|TS| - 9 = -lua
  0.30354|D|TS| - 10 = registerCoreModule('util/researchGE')
  0.30356|D|TS| - 11 = -userpath
  0.30359|D|TS| - 12 = C:\Users\lew_e\Documents\BeamNG.research
  0.37395|D|GFXInit::createDevice|Attempting to create GFX device 0: NVIDIA GeForce GTX 1080 (D3D11) [with output \\.\DISPLAY1]
  0.37397|D|GFXInit::createDevice|shader model: 5.0 , available video modes: 30
  0.37405|D|GFXInit::createDevice|Device created, setting adapter and enumerating modes
  0.46820|D|GFXD3D11Device::init|Hardware occlusion query detected: Yes
  0.46848|D|GFXCardProfiler::init|D3D11 -  (NVIDIA GeForce GTX 1080) [8079 MB] ver 
  0.46851|D|GFXCardProfiler::setCapability|GFXCardProfiler (D3D11) - Setting capability 'maxTextureWidth' to 8192.
  0.46854|D|GFXCardProfiler::setCapability|GFXCardProfiler (D3D11) - Setting capability 'maxTextureHeight' to 8192.
  0.46856|D|GFXCardProfiler::setCapability|GFXCardProfiler (D3D11) - Setting capability 'maxTextureSize' to 8192.
  0.46858|D|GFXCardProfiler::setCapability|GFXCardProfiler (D3D11) - Setting capability 'autoMipMapLevel' to 1.
  0.46861|D|GFXCardProfiler::setCapability|GFXCardProfiler (D3D11) - Setting capability 'lerpDetailBlend' to 1.
  0.46863|D|GFXCardProfiler::setCapability|GFXCardProfiler (D3D11) - Setting capability 'fourStageDetailBlend' to 1.
  0.46992|D|GFXD3D11Device::init|GFXDevice 11 - SM 5.000000
  0.47046|D|GFXD3D11WindowTarget::init|setting video mode: Width 1280, Height 720, Format 1, Windowed 0
[... 413 lines cut ...]
  size = number: 1.04858e+06
[... cut ...]
  Local variables:
  func = string: "onPreRender"
  funcList = table: 0x0217eadf4608  {1:function: 0x021777a38df0, 2:function: 0x0217643ea470, 3:function: 0x021764751c60 (more...)}
  (for index) = number: 1
  (for limit) = number: 11
  (for step) = number: 1
  i = number: 1
(5) Lua function 'luaPreRender' at file 'lua/ge/main.lua:298' (best guess)
  Local variables:
  dtReal = number: 0.046593
  dtSim = number: 0.0165
  dtRaw = number: 2.02987
  updateUIData = boolean: true
--------------- << END <<
```

pascale-commits, 2020-11-12:
Thanks! Is this a problem that is specific to your setup or do others using this build experience the same issue?

Can you post the output for typing into the game console `for k,v in pairs(Engine) do print(k) end`? Make sure that the `GE - Lua` option is selected in the drop down menu in the lower left. (You get to the console by using the '~' sign on your keyboard, but that might vary if you don't have an US layout.)

EthanJamesLew, 2020-11-13:
I believe that @kris-agile-hardware was able to see the same errors on his side. Here's the output:

```
 57.50659|I|GELua.ui_console.exec|GELua(Sandboxed) < "for k, v in pairs(Engine) do print(k) end"
 57.50768|I|GELua.ui_console.exec|output > "sceneGetCameraFrustum"
 57.50867|I|GELua.ui_console.exec|output > "Render"
 57.50964|I|GELua.ui_console.exec|output > "enableRenderMonitor3"
 57.51060|I|GELua.ui_console.exec|output > "destroyPhysics"
 57.51156|I|GELua.ui_console.exec|output > "getStartingArgs"
 57.51259|I|GELua.ui_console.exec|output > "___setters"
 57.51358|I|GELua.ui_console.exec|output > "getFrameLog"
 57.51455|I|GELua.ui_console.exec|output > "___getters"
 57.51558|I|GELua.ui_console.exec|output > "getCollisionDebugData"
 57.51661|I|GELua.ui_console.exec|output > "__index"
 57.51763|I|GELua.ui_console.exec|output > "Profiler"
 57.51860|I|GELua.ui_console.exec|output > "getStaticColDebugEnabled"
 57.52126|I|GELua.ui_console.exec|output > "createPhysics"
 57.52253|I|GELua.ui_console.exec|output > "Annotation"
 57.52355|I|GELua.ui_console.exec|output > "___type"
 57.52456|I|GELua.ui_console.exec|output > "castRay"
 57.52558|I|GELua.ui_console.exec|output > "Audio"
 57.52658|I|GELua.ui_console.exec|output > "___module"
 57.52758|I|GELua.ui_console.exec|output > "__newindex"
 57.52862|I|GELua.ui_console.exec|output > "getImageData"
 57.52958|I|GELua.ui_console.exec|output > "getEditorEnabled"
 57.53054|I|GELua.ui_console.exec|output > "Online"
 57.53120|I|GELua.ui_console.exec|output > "setStaticColDebugEnabled"
 57.53249|I|GELua.ui_console.exec|output > "getStartingCommandLine"
 57.53333|I|GELua.ui_console.exec|output > "Platform"
 57.53418|I|GELua.ui_console.exec|output > "Debug"
```

Palculator, 2020-11-13:
Hey, I unfortunately haven't been able to reproduce this on my end either. I've been testing with the same build you seem to be running:

```
-- Log started - v 1.7.0.0.0 - x64 - build 0 - 2020-11-13 -- 09:32:14 -----
-- private build Marc - Tue Sep 15 11:16:02 2020
```

Using a license file from the same registration form you used. What's peculiar is that, despite the license check itself clearly working because the game complains about a missing or invalid license for you, this log line that I find in my log is not present for you:

```
  0.05929|I|verifyResearchLicense|BeamNG.research license verified.
```
Can you maybe email me your license file to mmueller (at) beamng.gmbh? Maybe that will allow me to reproduce the issue as my own test license verifies without problem. Thanks!

EthanJamesLew, 2020-11-13:
Emailed! Thanks

Palculator, 2020-11-17:
Thanks, I was able to replicate and fix the issue. There should be a new zip available for download at the link you've been emailed after registering. It's labeled version 1.7.0.1. I've tested it with your license and got it working fine. I hope the same goes for you!

EthanJamesLew, 2020-11-17:
@Palculator this fixes the issue for me. Thank you!

CC @kris-agile-hardware

closed 2020-11-17 by EthanJamesLew