BeamNG/BeamNGpy #86: Create a path

ASionville, 2020-11-08:
Hi everyone !

I would like my car to automatically create a path for an ai.
I there a way to say something like "create path from here to {enter point}" ?

I do not want the ai to go there by itself, I just want to get back the nodes coordinates of the path.

Is it possible ?

pascale-commits, 2020-11-11:
Hi,

so in short: yes this functionality is available on the lua side of things but it's not exposed through the python API. For now its not planned to expose this function through BeamNGpy. 

But we will provide you the lua code, so that you can write your own extension :)

Some notes on the following lua code:
- the function you can use computes a path based on "nodes" in the map, they are different from a position vector 
- it is possible to get the closest nodes that define a road segment from any given position, but since there are two of them, one might be "behind" the vehicle 
- the following code runs in the vehicle VM, not in the game engine
- if you are not sure on how to approach writing an extension, see this tutorial on [how to connect your own lua code to BeamNGpy](https://github.com/BeamNG/BeamNGpy/blob/dev/examples/modInterface/usingmodsWithBNGpy.ipynb) (be sure to check out the dev version, the one in master has no explanation)

```
mapmgr.requestMap() --initialize the map (if you initialized the AI beforehand you might not need this line)
startPos = vec3(obj:getPosition()) -- get current vehicle position as starting point
goalPos = vec3(156, 756, 123) -- some random values that represent the goal
startNodeA, startNodeB = mapmgr.findClosestRoad(startPos) -- returns two node names 
goalNodeA, goalNodeB = mapmgr.findClosestRoad(startPos) 
nodePath = mapmgr.getPath(startNode, goalNode, 0, 5, 8, 8) -- this returns a list of path nodes aka strings
print(mapmgr.map.nodes[nodePath [1]].pos) -- this contains the coordinates for the first node in the path
```

ASionville, 2020-11-11:
Ok, thanks!

I think I will compute the vectors vehicle-point_1 and vehicle-point_2 to find out wich one is in front of the car

Just a thing: what is the "vehicle VM"?

pascale-commits, 2020-11-12:
Lua uses two types of virtual machines. One for the game engine and one for the vehicles. Each vehicle has its own VM. 
That means that each vehicle loads its own copy of the code in /path/to/beamng/lua/vehicles.
It sounds like you've never used the game's command line before. In game you can open up the console and type in lua code and execute it (only if its a valid chunk though).
To the left of the input field you can choose where to execute it (ge or any spawned vehicle).

pascale-commits, 2020-11-12:
Also, the values in the `getPath` function are random, this part of the code is documented though, so take a look at it and set the values as needed.    
Although the function is called through the `mapmgr` module, the function and the documentation can be found in .../ge/map.lua.

ASionville, 2020-11-12:
Ok, thanks for everything, I will learn LUA to make the extension

ASionville, 2020-11-14:
Sorry to reopen the issue but there are a few questions about the extension :
 - Is the code that you sent above enough to write the 1st point coordinates ?
 - In what folder should I put the file ?
 - How should I call it, main.lua ?

pascale-commits, 2020-11-16:
Hi, I don't understand your first question. But I think you might find your answer if you run the code in the game's command line. It should appear if you use the key with the `~`, but that might vary, depending on your keyboard settings. If you need to know the content of a variable, then call `dump(someVar)`. If you want to know what functions an object has then use `getmetatable(someObject)`. If you changed your code and want to reload lua (without restarting the game) on the ge side use `Ctrl+L`, for the ve side of your current vehicle use `Ctrl+R`.

You can put it in the game's directory: `/lua/vehicle/extension/<yourModName>.lua`. Then you will also have to add another line in the `/lua/vehicle/main.lua`: `extensions.load('yourModName')` and it is then available through `extensions.yourModName`. I encourage you to go through the tutorial once more and read though its explanations :)

ASionville, 2020-11-21:
I get an error when I start Beamng with the extension

The car doesn't spawn and I got an error in the console :

I have put `extensions.load("autoTravel")` at the end of` `/lua/vehicle/main.lua`
I have create a file `/lua/vehicle/extension/autoTravel.lua` with this inside :

`-- This Source Code Form is subject to the terms of the bCDDL, v. 1.1.
-- If a copy of the bCDDL was not distributed with this
-- file, You can obtain one at http://beamng.com/bCDDL-1.1.txt
local M = {}

mapmgr.requestMap() -- On demande les données GPS

local startPos = vec3(obj:getPosition()) -- On recupere la position actuelle
local goalPos = vec3(156, 756, 123) -- On met un objectif (coords aléatoires)

local startNodeA, startNodeB = mapmgr.findClosestRoad(startPos) -- Les deux points les plus proches du depart
local goalNodeA, goalNodeB = mapmgr.findClosestRoad(goalPos) -- Idem pour l'arrivee

local nodePath = mapmgr.getPath(startNodeA, goalNodeA, 0, 5, 8, 8) -- Path = Ligne (Ici le trajet complet)

local pathLength = table.getn(nodePath)
print(mapmgr.map.nodes[nodePath [i]].pos)
--for i=1,pathLength do
--	print(mapmgr.map.nodes[nodePath [i]].pos)

return M`

I have first added in M variable because I thought I seen that all the other files have it
It appears that the problem is with the line 3 (copy of the error in the console) :

20.41164|E|libbeamng.lua.V|*** FATAL LUA ERROR: lua/vehicle/extensions//autoTravel.lua:6: attempt to index global 'mapmgr' (a nil value)

Did I miss something out ?

pascale-commits, 2020-12-03:
Hi, excuse the delay.

I found that this code will only work reliably if used in scenarios.
Also, you seemingly need to separate the mapmgr initialization from the path planning.

```lua
-- put every public attribute and function into this table
local M = {}

local _log = log
local function log(level, msg)
    _log(level, "issue86", msg)
end

local function onExtensionLoaded()
    log("D", "init")
    mapmgr.requestMap() 
end

local function foo()
    local startPos = vec3(obj:getPosition()) 
    local goalPos = startPos + vec3(100, 0,0) 
    local startNodeA, startNodeB = mapmgr.findClosestRoad(startPos) 
    local goalNodeA, goalNodeB = mapmgr.findClosestRoad(goalPos) 

    local nodePath = mapmgr.getPath(startNodeA, goalNodeA, 0, 5, 8, 8) 
    dump(nodePath)
    local pathLength = table.getn(nodePath)
end

-- public interface

-- hook called every time the extension is loaded
-- for other available hooks take a look at other vehicle extensions 
M.onExtensionLoaded = onExtensionLoaded 

M.foo = foo

return M
```

If you put the `extensions.load("autoTravel")` at the end of the main, then the `mapmgr` has not been loaded yet. Its loaded in the `init(...)` function of the `main.lua`.

Let me know if there are other questions.

ASionville, 2020-12-03:
Thank you for the code !

A question that comes immediately : What does the 0 5 and the two 8 stand for in the getPath function ?

pascale-commits, 2020-12-04:
The 5,8,8 are not used, I've removed it from the example :-)

The third function argument of the `mapmgr.getPath` function, `dirMult`, is used as a path computation penalty. 
If you don't care about the intended direction of the lane you are driving on, then set it to 1, if you want to use only the correct side of the road, then use 1000.

ASionville, 2020-12-07:
Ok, seems to work as I don't get any error message anymore

Now, how could I start the foo() function from the console or (better) from python ?

pascale-commits, 2020-12-08:
How to integrate a lua function into the python API is described in the tutorial that I've posted above ;-)

From the console you can call it by typing `extensions.bar.foo(args)`, given that you've added `foo` to the module interface.
If its in a subdir then it would be `extensions.subdir_bar.foo(args)`.

closed 2020-12-08 by pascale-commits

lsm2842035890, 2024-03-25:
[image]
[image]
hello friends~~I loaded the gridsmall map and paved a simple road. I want the vehicle to run according to my trajectory. Why does the vehicle not move?