BeamNG/BeamNGpy #89: Get values from LUA function

ASionville, 2020-12-08:
Hello,

A little thing I am messing with:

I have my LUA code working properly and returning the right output if I type into the ingame console, but I can't figure out how to get the same result with python.

I've seen [that](https://github.com/BeamNG/BeamNGpy/blob/dev/examples/modInterface/usingmodsWithBNGpy.ipynb), but I do not know what to put in the dictionary in the 5th part. Also, it seems to be a 'Bar' handler, but I'm not sure.

Here is the code (trunk/lua/vehicle/extensions/autoTravel.lua):
```-- put every public attribute and function into this table
local M = {}

local _log = log
local function log(level, msg)
    _log(level, "autoTravel", msg)
end

local function onExtensionLoaded()
    log("D", "init")
    mapmgr.requestMap() 
end

local function foo()
    local startPos = vec3(obj:getPosition()) 
    local goalPos = startPos + vec3(100, 0,0) 
    local startNodeA, startNodeB = mapmgr.findClosestRoad(startPos) 
    local goalNodeA, goalNodeB = mapmgr.findClosestRoad(goalPos) 

    local nodePath = mapmgr.getPath(startNodeA, goalNodeA, 0) 
    dump(nodePath)
    local pathLength = table.getn(nodePath)
end

-- public interface

-- hook called every time the extension is loaded
-- for other available hooks take a look at other vehicle extensions 
M.onExtensionLoaded = onExtensionLoaded 

M.foo = foo

return M```

In I put `extensions=["autoTravel"]` when I create the vehicle in python, but when I do this:
```data = dict(type='Bar')
print(vehicle.send(data))```
it always returns None

What did I do wrong?

pascale-commits, 2021-01-04:
Hello, I've updated the tutorial to include an example on how to do that.

closed 2021-01-04 by pascale-commits

pascale-commits, 2021-01-05:
Hi! I've added how to do it to the example. Since communication between the two processes is handled through a socket, the user need to implement any data transfer her(him)self.