BeamNG/BeamNGpy #95: Software does not start and fail silently when home folder contains space

alessiogambi, 2021-01-11:
Hi, I am not sure if this is mentioned anywhere, but with 1.7.0.1 we could not run BeamNG.research when the user path has an empty space.

We figured this out after some trial and error (from the console/command line): we manually started BeamNG.research by adding and removing the usual options (-console, -rport, etc). Removing the -userpath one made it work.

Maye there's a way to handle this at the API level?

pascale-commits, 2021-02-01:
Thanks for the report. We are working on it.

Another way of using the simulation with manual startup is to call `beamng.open('launch=False')`.

alessiogambi, 2021-02-01:
FYI. We resorted to let user change the location of beamng user folder.

Palculator, 2021-03-05:
I thought this was a problem in BeamNGpy but it turns out the failure is actually in BeamNG.tech. I've added a note about it to the README and moved the issue to our internal BeamNG.tech issue tracking. Since it's not really related to BeamNGpy I'll close this issue.

closed 2021-03-05 by Palculator