BeamNG/BeamNGpy #102: Pixel-Depth Scaling Incorrect

BleedObsidian, 2021-02-05:
I believe the calculation for the pixel-depth of the Camera sensor to be incorrect. Please see the below two images:

[image]

As you can see, the image on the left is using the default `near_far=(0.01, 1000)` during Camera initialisation. However this image is completely useless for vehicle navigation because all objects close to the car for a significant distance, such as bushes, have a lightness value of 0. So, I have reduced the Camera on the right's `near_far=(0.01, 100)`, expecting that objects at a distance of 100 to have a lightness value of 255 and objects at 0.01 to have a lightness value of 0. This should result in objects closer to the vehicle having a much more fine and resolved lightness scale. As you can see this is not the case.

The lightness values are not scaling whatsoever with the `near_far` setting of the camera. The image on the right shows that all objects within range of the cameras `near_far` clipping range, have the exact same lightness values as the image on the left. The only difference is the sky appearing darker because it is clipped at a shorter distance. I am almost certain this is not correct behaviour?

[image]

I believe the above code located in the [Camera](https://beamngpy.readthedocs.io/en/latest/_modules/beamngpy/sensors.html#Camera) class in the `decode_response` method is maybe the culprit. 

If indeed this is normal behaviour, how can I go about making objects closer to the vehicle have a finer resolution of depth? Because these depth images are completely useless to me.

BleedObsidian, 2021-02-05:
To reproduce this behaviour, simply attach a [Camera](https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.sensors.Camera) to a vehicle and adjust the `near_far` setting:

[image]

mihalykorodi, 2021-02-13:
Hi!
I have the same problem. This should fix it #105

BleedObsidian, 2021-02-13:
I can confirm this PR does partially fix the issue, the below image is for `near_far=(0.01, 100)` :

[image]

So thank you for that @mihalykorodi 

However, behaviour is still quite peculiar when playing with the near part of parameter. The below image is taken with `near_far=(10, 100)` :

[image]

This raises the question that actually the `near_far` parameter should only address camera clipping, and another parameter should address the minimum and maximum distance thresholds for white and black pixels. In the above image, I would expect the sky to still remain white, and the nearest objects to be completely (they are currently a mid grey in this example). The white object in the bottom left of this image is actually a result of seeing 'under' the map and viewing a far away downhill slope.

Like i said, I think camera clipping and white/black thresholds should become two separate parameters.

BleedObsidian, 2021-02-14:
I've implemented this functionality using np.interp, I will soon be posting a pull request.

BleedObsidian, 2021-02-14:
@mihalykorodi Please see the pull request that I have made (#106) that incorporates your fix as well as handling the near setting more suitably. Your fix does not work appropriately when changing the minimum part of the `near_far` tuple.

Also, your fix does not allow a minimum distance without clipping the camera either like so:

[image]

mihalykorodi, 2021-02-14:
Thank you for the info You are right! I did not check the near part at all.

pascale-commits, 2021-02-15:
Hi, thanks for your contribution! The current submission format does not fit our workflow, so I will set up a guideline on how to contribute to our project. If you are willing to change your contribution format according to our guidelines, then we are happy to integrate it into our project. Please wait until further notice before adding any more changes. If you do not find the time to do it, we will of course fix the bug.

BleedObsidian, 2021-02-15:
@pascale-commits Yeah I'm happy to change to your workflow, send me the guideline when you can. I'm guessing you'd rather me create a seperate branch and create a PR to the dev branch?

BleedObsidian, 2021-02-17:
@pascale-commits I've updated the PR to merge to dev instead of master if that's all your looking for.

pascale-commits, 2021-02-18:
Thanks, that is part of it :)

pascale-commits, 2021-03-03:
Hi :)

Thanks for your patience! We took the opportunity to set up some contribution guidelines for our repository [here](https://github.com/BeamNG/BeamNGpy/blob/dev/docs/contributing.md). This includes that users need to agree to our CLA before we can accept any pull requests. This is some unexpected administrative overhead but we hope this doesn't discourage anyone from contributing to our project.

BleedObsidian, 2021-03-07:
@pascale-commits Thanks for the guidelines. I've submitted a new pull request and also agreed to the CLA (sent an email to info@beamng.gmbh) https://github.com/BeamNG/BeamNGpy/pull/110