BeamNG/BeamNGpy #109: Unable to detect Beam Binary

vNemesis, 2021-03-07:
Hi,

I am attempting to use the example provided in the readme but i keep getting this error

[image]

Any ideas as to what the issue could be?

I have tried setting the home path to the folder contain the binary in several different ways but all produce the same error

pascale-commits, 2021-03-10:
Its most probably, because home needs to point to the root folder, and not to the `bin64` one.
Just change your code to `BeamNGpy('localhost', 64255, home='../program')`.

vNemesis, 2021-03-12:
I tried but still get the same error : (

I also get and error about my user dir having a space in the name but followed the instructions to manually set the path. i even set it to a path with no spaces for the key file.

I should mention i am using Anaconda Navigator but have set up a separate environment with the necessary modules  and attempt to run the script file using the PowerShell prompt option in the navigator. not sure if this would cause issues?

vNemesis, 2021-03-12:
I found out the issue, didn't realise is resolved the user path itself instead of using the startup.ini in the beam folder.

I set the path to my custom location and it worked : )

closed 2021-03-12 by vNemesis