BeamNG/BeamNGpy #111: Polling sensors hangs randomly

mihalykorodi, 2021-03-12:
Hi!

Calling vehicle.poll_sensors() hangs randomly. It can't be reproduced intentionally, it happens randomly, around 7 out of 10 start, without changing anything. There is no error in the BeamNG console or in the  BeamNGpy log.

I have cleared all caches, and reinstalled everything (BeamNG.tech, BeamNGpy). 

I will try removing sensors one by one, as soon as I have time.

Some info:
BeamNG.tech version: 0.21.3.0
BeamNGpy version 1.19.1

pascale-commits, 2021-03-15:
Hi, good catch! I will try it out as soon as I get around to it, but that might take some time (~2 weeks).

mihalykorodi, 2021-03-15:
Hi!

I have another issue, what is related to multiple cameras when they have different range/near far values. I mentioning this here because maybe it should be taken into consideration at implementation. I will write that issue as soon as possible. In a nutshell sometimes a "water wall" gets rendered on the farther camera's picture.

I will edit this comment or write a new when the issue is created.

pascale-commits, 2021-04-28:
Hi,
this took a bit to get around to it. If you try out the latest dev version, it should be fixed, otherwise its going to be included into the BeamNGpy version :)

closed 2021-04-28 by pascale-commits