BeamNG/BeamNGpy #115: Camera polling very slow since 1.19.1

BleedObsidian, 2021-03-22:
Attaching a camera sensor that returns colour images to a car causes the simulator to become extremely laggy and low-framerate. 

The call to `vehicle.poll_sensors()` takes so long that it becomes impossible to achieve more than a few fps. You don't even need to try and render the image, just simply attach a camera to the car and observe the terrible lag that occurs within the simulator itself.

This has only happened since the latest 1.19.1 release and the rebrand to BeamNG.tech

BleedObsidian, 2021-03-22:
Just confirmed, I've downgraded my copy of BeamNG.research to 1.7.0.1 and BeamNGPy to 1.17.1 and the camera works great again! I wonder if shmem=True (by default) is not actually causing it to use shared memory at all, hence the lag?

I also noticed that the camera is now properly mounted to the car (the latest version the camera seems to move relative to the body of the car as the suspension travels). I was also previously (in the latest version) getting some strange shadow artifacts, which have now completely gone.

I think the camera needs relooking at in the latest version, it's got some significant bugs compared to 1.17.

pascale-commits, 2021-07-28:
@Palculator fixed this issue. It seems that the annotations where always rendered alongside the color images, which should be resolved in newer versions.

closed 2021-07-28 by pascale-commits