BeamNG/BeamNGpy #118: Set waypoints requires a string

hildebrandt-carl, 2021-05-01:
Hi,

I want the vehicle to navigate to a series of waypoints on the map autonomously. I noticed that there is a function, `ai_set_waypoint`, that is designed for this purpose. However, this function takes in a string (for example, `Bridge4_B`), which I assume represents some waypoint on the map.  The problem is I can't find where these strings are defined, nor what positions they represent.

Where are these strings defined? Also, is there a way to send this function waypoints defined as a tuple or array?

Thanks!

Palculator, 2021-05-05:
Hello! The `Scenario` class contains a method to download the scenario's scene tree. While I should add better methods to query that tree for specific objects, in the meantime you can traverse it for objects of a certain type -- in your case `BeamNGWaypoint`. It would look something like this:

```python
scenario.sync_scene()
def get_waypoints(node, waypoints):
    if node.type == 'BeamNGWaypoint':
        waypoints.add(node)
    if node.children:
        for child in node.children:
            get_waypoints(child, waypoints)

waypoints = set()
get_waypoints(scenario.scene, waypoints)
```
The `sync_scene()` function is a bit slow at this point (this and related features are kind of WIP) but seeing as the waypoints are unlikely to move around, you can get away with synchronizing it once after loading a scenario. Documentation here: https://beamngpy.readthedocs.io/en/latest/api/beamngpy.html#beamngpy.scenario.Scenario.sync_scene

hildebrandt-carl, 2021-05-05:
Hi @Palculator, thank you so much for the answer. The `sync_scene()` function is indeed a bit slow and takes over a minute to complete on some of the larger maps like `west_coast_usa`. The only other thing I would add to your answer is for anyone looking at this issue in the future, you can then access the waypoint locations and names using:

```
scenario.sync_scene()
waypoints = set()
get_waypoints(scenario.scene, waypoints)
for w in waypoints:
    print(w.name)
    print(w.pos)
```

I understand most of these functions are a WIP, but regardless I know there are many researchers out there all over the world that thank you for making your software available :)  So thank you a ton!

closed 2021-05-05 by hildebrandt-carl

Palculator, 2021-05-05:
Oof, well that's certainly too slow. I think it's due to the fact that Python right now first gets list of objects to sync and then gets their properties one object after another, only one being handled per frame. I'll optimize that in the near future.

ShenShenRuYe, 2024-09-29:
Where is this function get_ waypoints ?

aivora-beamng, 2024-09-30:
> Where is this function get_ waypoints ?

Hi, it's actually called `find_waypoints`, you can find the docs here:
https://beamngpy.readthedocs.io/en/latest/beamngpy.html#beamngpy.Scenario.find_waypoints