BeamNG/BeamNGpy #121: IMU Unit System

danielstuart14, 2021-05-05:
Hi, I'm currently doing a research using beamNG.tech and one of the sensors I'm using is the IMU provided.
I need for my research the linear acceleration values of the vehicle.

I'm having a little bit of trouble understanding the IMU values, as they are not very well documented.

These are my questions:
a) Which unit system is the IMU reporting? (rad/s for rotation? Newtons for linear force?)
b) The force reported uses the mass of the vehicle or it just represents the acceleration? (If so, is it reporting m/s^2?)

Thank you,

Palculator, 2021-08-05:
Hey, the IMUs do report rotation in rad/s and acceleration in m/s^2. This does not include gravity. The current value of gravity can be obtained in Lua using `core_environment.getGravity()`, but I can expose that to BeamNGpy with relative ease (I actually exposed `setGravity()` but I guess never did the same for the getter 🙃.)

What your second comment points out does seem like a bug though; I'll change the IMUs to rotate according to the orientation of the vehicle they are mounted to.

jvalentine, 2021-09-25:
Was this resolved? Does the IMU now relate to the vehicle's orientation?