BeamNG/BeamNGpy #123: LIDAR example fails to run, Assertion error

vNemesis, 2021-05-16:
Hi,

I am attempting to run the LIDAR tour example but i get this error when the scenario loads.

[image]

It seems an assertion is failing when running the poll_sensors method.

closed 2021-05-26 by vNemesis

nebriv, 2021-06-03:
@vNemesis I'm also running into this issue - I see you closed it, did you find a solution?

vNemesis, 2021-06-07:
@nebriv No sorry, I no longer needed to use it. I did find that there was a similar issue when using a camera. it seems to be caused when using the step part of the code.

[image]

Some combination of removing these these lines stopped the issue when using a camera, not sure if it works the for the LIDAR as well sorry!